dbuezas/lgt8fx #168: MD-328D(LGT8F328D) SPI: compile error

jg1uaa, 2021-06-06:
When building sketch that uses SPI interface for MD-328D, compile error (undefined SPFR) occurs.
There is no SPFR in lgt8f/cores/lgt8f/lgtx8e.h and we can avoid this issue adding same definitions from lgt8xd.h.

I think lgtx8e.h needs to be revised. Does anyone who can work E2PROM and any other peripherals properlly?

LaZsolt, 2021-06-06:
My board working with 328D. (Dallas temperature sensor + ENC 28J60 SPI ethernet adapter + Webserver)
[image]

LaZsolt, 2021-06-06:
Ahh. I'm not remembered, my sckech not include SPI library.

jg1uaa, 2021-06-09:
Wow, this red board looks cool. where did you buy it?

BTW, SPI.cpp compile error occurs;
- SPIClass::transfer(void *buf, void *retbuf, size_t count) at SPI.cpp is compiled when __LGT8F__ defined
- it requires SPFR (and its bitfield) definition
- only cores/lgt8f/lgtx8p.h has the definition

So simply adding it to cores/lgt8f/lgtx8e.h is a solution. but SPIClass::transfer(void *buf, void *retbuf, size_t count) supports for LGT8F328P, not D.
In SPI.h, only LGT8F328P has special code... Is this care required for D variant or not? This is difficult to judge.

jg1uaa, 2021-06-09:
MassDuino's package does not have SPIClass::transfer(void *buf, void *retbuf, size_t count) at SPI.cpp and SPI.h.

I don't know why this method is implemented and used for. I think this should be limited for __LGT8FX8P__, not for __LGT8F__.

LaZsolt, 2021-06-09:
> Wow, this red board looks cool. where did you buy it?

https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20210609104247&SearchText=lgt8F328d

> Is this care required for D variant or not?

To use SPI interface of LGT 328D variant, the SPI.h of Arduino package working perfect.
Soo to using SPI of 328D, no need to use this dbuezas/lgt8fx package's SPI.h.
I dont know how to 'tell' it to the compiler.

jg1uaa, 2021-06-09:
D variant defines LGT8FX8E and LGT8F symbol, so #if defined (LGT8F) ~ #endif is compiled.
it is easy to limit SPIClass::transfer for P variant, but adding source from Arduino for D variant is required...?

LaZsolt, 2021-06-09:
> adding source from Arduino for D variant is required

Yes, this could be a solution.

jg1uaa, 2021-06-09:
I just ordered WEMOS XI, thanks for inforation about this board and bootloader.

jg1uaa, 2021-06-10:
Hmm, this SPI problem comes from PR #45 , revise needed.

jayzakk, 2021-06-10:
Yes, this special performance function should be limited to the P variant, so the ifdef has to change :/ 
The D variant has to be treated equal to a real Atmega: no such function.

dwillmore, 2023-01-12:
@jg1uaa Is this issue resolved by this commit?  Can you test wil current 1.0.7 release?

jg1uaa, 2023-02-10:
Sorry, I checked only building process was no problem. Checking with real hardware is not yet.

LaZsolt, 2023-05-21:
> Sorry, I checked only building process was no problem. Checking with real hardware is not yet. But I think this issue can be closed.

There were no problem reported until now so I've closed this issue.

closed 2023-05-21 by LaZsolt