dbuezas/lgt8fx #173: Can't compile sketches with SoftwareSerial anymore

seisfeld, 2021-07-05:
Somehow #40 breaks SoftwareSerial support for me. I get similar error message as @LaZsolt gets here: https://github.com/dbuezas/lgt8fx/pull/40#issuecomment-778845698. Last good version of pins_arduino.h for me is from commit 6a5249947ed0ed6ed60ced8007ef3f25c248085f.

LaZsolt, 2021-07-05:
The newest **pins_arduino.h** available here: https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/variants/standard/pins_arduino.h
Try to install it!

seisfeld, 2021-07-05:
I am using that version, well I wanted. That's what I'm saying. When I go back in the git history of that file, 6a52499 is the first working version for me.

LaZsolt, 2021-07-05:
I tried to compile `SoftwareSerialExample.ino` but no errors at all, even I set the compiler warnings to max level. (All warnings)

seisfeld, 2021-07-05:
You are right @LaZsolt. Thanks for checking! Somehow the Arduino IDE got messed up with different versions of the core. Cloning the repo fresh resolved the issue. Sorry for the confusion. Closing...

closed 2021-07-05 by seisfeld