dbuezas/lgt8fx #174: The mysteries about PE4 PE5 PE6

seisfeld, 2021-07-05:
I want to use PE4 and PE5 in my sketch (knowing that these are not exposed on regular LGT boards with LQFP32 package, but I have a custom board). After @jayzakk made PE6 available as A10 via #36 I was curious if one could also use PE4 and PE5 even without "official" Ax naming etc. Checking `pin_arduino.h` everything should be there. Here is what I did:

```
const uint8_t testPin = XX; // XX = E4, E5 or E6

void setup() {
  pinMode(testPin, INPUT_PULLUP);
  Serial.begin(115200);
  Serial.println(testPin);
}

void loop() {
  Serial.println(digitalRead(testPin));
  delay(1000);
}
```

Running this the console prints the internal "port number" (?) and then continuously prints 1, when shorting the relevant pins to GND I should see the console printing 0. So far so good.

If compile it with the LQFP32 option I get the following:

E4 (prints 24), works
E5 (prints 26), doest not work
E6 (prints 25), works but not on AREF pin but on PE5 (huh?!)
A10 is equivalent to E6

On the other hand, if I compile it with LQFP48 get the following:

E4 (prints 31), works
E5 (prints 32), works
E6 (prints 25), works (on AREF as expected)
A10 is equivalent to E6

Can anyone shed some light on this? Despite the different "port number" (?) when printing the bit field (`_BV()`) points to the same bits on the PORTE in `pin_arduino.h`. So from my understanding this should work with LQFP32 setting as well, but it does not. Please help. :)

PS: The analog read on A10/AREF (see #36) works with both, it's just when you want to use these as GPIOs something looks mixed up.

LaZsolt, 2021-07-11:
It's really mysterious. I'll try solve it, both with 328P and 328D when I have more free time.

LaZsolt, 2021-07-17:
Not yet tested, but

1. To use of PE6 as GPIO must set a bit:
[image]

2. I think in `pins_arduino.h` line 185 and line 186 are exchanged. I think this is correct :
´´´
#define E5	25
#define E6	26
´´´

seisfeld, 2021-07-17:
Dunno if this helps, but with direct port manipulation I can read in E4 and E5 just fine when using the LQFP32 option:

```
// E4 and E5 input
bitClear(DDRE, 4);
bitClear(DDRE, 5);

// E4 and E5 pull up
bitSet(PORTE, 4);
bitSet(PORTE, 5);

Serial.println(bitRead(PINE, 4));
Serial.println(bitRead(PINE, 5));
```

LaZsolt, 2021-07-17:
My first thought about of error was not good above because:
> 1. `digitalRead()` sets the required bit. So no need to to pay attention about PMX2.
> 2. There was no errors with lines 185 and 186 in `pins_arduino.h`.

But I found a typing error in merge request https://github.com/dbuezas/lgt8fx/pull/36
If you watch the modified files https://github.com/dbuezas/lgt8fx/pull/36/files , you can see in the last modified lines in `pins_arduino.h`

digital_pin_to_bit_mask_PGM[] section
```
	_BV(5), /* 26, port E5 */
	_BV(6), /* 25, port E6 */
``` 
only the remark was changed not the bit field.

seisfeld, 2021-07-17:
There is more typos a few lines before in the the comments:

```
	_BV(2), /* 34, port F2 */
	_BV(3), /* 35, port F3 */
	_BV(4), /* 34, port F4 */
	_BV(5), /* 34, port F5 */
	_BV(6), /* 34, port F6 */
	_BV(7), /* 34, port F7 */
```

This should count up to 39, shouldn't it?

LaZsolt, 2021-07-17:
Yes, a beauty bug. :)

LaZsolt, 2021-07-17:
Tested with `digitalRead()` and `digitalWrite()`. The result is good, both with 328P and 328D, after port E5 and E6 bitfield is corrected in `pins_arduino.h`
```
391 #else
392	_BV(0), /* 22, port E0 */
393	_BV(2), /* 23, port E2 */ 
394	_BV(4), /* 24, port E4 */
395	_BV(6), /* 25, port E6 <--- This two */
396	_BV(5), /* 26, port E5 <---  lines   */
397	_BV(6), /* 27, port C6 */
398 #endif
399 #endif
```
But in case of 328D, the lgt8fx package not disabling AREF when setting pin mode for E6, and not disabling SWD + SWC when setting pin mode E0 or E2. It needs to be corrected later.

dwillmore, 2023-01-12:
Is this issue also resolved by #185 ?

LaZsolt, 2023-01-25:
This issue has been resolved with https://github.com/dbuezas/lgt8fx/pull/185 and board package release v2.0.0.
But read this: https://github.com/dbuezas/lgt8fx/discussions/242

closed 2023-01-25 by LaZsolt