dbuezas/lgt8fx #178: How to use PCINT4?

brother-yan, 2021-07-24:
Since mega328p doesn't have this interrupt vector. How to use this interrupt in lgt8f328p? Thanks!

LaZsolt, 2021-07-25:
I never use pin change interrupts. I found few examples for you:
https://thewanderingengineer.com/2014/08/11/arduino-pin-change-interrupts/
https://www.teachmemicro.com/arduino-interrupt-tutorial/
https://github.com/NicoHood/PinChangeInterrupt
I think it is quite simple. You need only change the names of ports or interrupt vector names.

dwillmore, 2023-01-12:
@brother-yan what was your results with LaZsolt's suggestions?

LaZsolt, 2023-01-26:
Since the @brother-yan hasn't loggen in the Github for two years, I close this issue.

closed 2023-01-26 by LaZsolt

jayzakk, 2023-01-26:
That's perfectly fine, I added them in #40

brother-yan, 2023-02-13:
@dwillmore I don't use LGT MCU since 2020. I have a solution:

添加文件：`IVT_extended.S`
```S
#include <avr/io.h>

#ifdef __AVR_MEGA__
  #define XJMP jmp
  #define XCALL call
#else
  #define XJMP rjmp
  #define XCALL rcall
#endif

.section .vectors, "ax", @progbits
  .weak __vector_26
  ;.set __vector_26, __bad_interrupt
  XJMP  __vector_26
  .weak __vector_27
  ;.set __vector_27, __bad_interrupt
  XJMP  __vector_27
  .weak __vector_28
  ;.set __vector_28, __bad_interrupt
  XJMP  __vector_28
  .weak __vector_29
  ;.set __vector_29, __bad_interrupt
  XJMP  __vector_29
```
即可

LaZsolt, 2023-02-14:
Thank you for the answer.-