dbuezas/lgt8fx #202: How to use sleep modes on LGT8F328P ?

Vickthorn, 2022-03-18:
Hi !

So I moved from Arduino Nano to the LGT8F328P and I feel a little bit lost. The LowPower.h library I was using is not working anymore.

I'm looking for another way to reduce consumption for a certain period of time (2 seconds). I already have the LGT file installed in my hardware arduino location.

Thanks for your help !

alxgarza, 2022-07-04:
I too have problems with the power consumption.  The lowest I can get is 170µA, nowhere near the 1µA@3.3V that is mention on the manual or that I can achieve with a atmega328.

LaZsolt, 2022-07-05:
I think in this lgt8fx package the sleep modes controlling software elements and examples are unmodified and not tested.

To set the LGT8F328P to (deep) sleep mode (and wake up settings) is much more complicated than in case of ATmega 328P because of
-LGT8F328 has more (deep) sleep control registers ( PRR1, DPS2R, IOCWK and more bits in MCUCR and MCUSR )
-it has more wake up modes/sources ( IO level change, watchdog, ADC, etc. )
-has more circuit design restriction ( eg. need pull-up or pull-down resistors connected to some IO-s to achieve lower power consumption)
-need to power down some more internal circuits than ATmega ( eg. low voltage detector (LVD) or internal voltage source circuits and more )

You can learn from databooks and untested examples :
https://github.com/dbuezas/lgt8fx/tree/master/lgt8f/libraries/PMU
https://github.com/dbuezas/lgt8fx/tree/master/lgt8f/libraries/WDT
https://github.com/dbuezas/lgt8fx/blob/master/docs/LGT8FX8P_databook_v1.0.5-English.pdf
https://github.com/dbuezas/lgt8fx/blob/master/docs/Porting_from_LGT8FX8D_to_LGT8FX8P_v1.0.0.pdf
( the `lgt8f328p_spec.h` is integrated into this package so you don't have to include it)

Vickthorn, 2022-07-05:
Hi everyone,

Well 170μA seems already low hahaha, with the 5v pin I obtained 1mA for a
project with the purple LGT8F328P board, I was happy about it. But I tried
the same with the green board, and I obtained 25mA, impossible to get lower
values. I used destroyed led technique, no 5v regulation, and deep sleep
mode code. The green and purple boards seems to have a different PCB
layout...

For the deepsleep mode I directly use the internal sleep register as the
lowpower library is not compatible with this boards.

LaZsolt, 2022-07-05:
The purple and the green boards has USB to serial chip CH340 or HT42B534 ( and AMS1117 LDO regulator ) which has some current load.

alxgarza, 2022-07-07:
> I think in this lgt8fx package the sleep modes controlling software elements and examples are unmodified and not tested.
>
> To set the LGT8F328P to (deep) sleep mode (and wake up settings) is much [...]

Has anyone achieved the 1µA in standby mode?  (This is easily achieved on a a atmega328p)

Running the same code

dwillmore, 2023-01-12:
Are any of the interested parties still involved in this or has this issue passed from usefullness?

LaZsolt, 2023-02-06:
@alxgarza 
> Has anyone achieved the 1µA in standby mode?

I tested and reached current lower than 2 µA in deep-sleep mode. (Measured with a 3 and 1/2 digits METEX multimeter.)
I was used the SSOP20 Mini style board for testing because only three components had to be removed. The voltage regulator, and 2 LED's resistors. (The LQFP32 Mini style board can be just as good.)
At the first test I didn't remove the builtin LED's resistor so it turned out the pull-up resistor switched on when the deep-sleep mode started. After this I removed this resistor and connected an external LED with a 10k resistor between D13 Arduino pin and VCC, because when pull-up resistors switching on this LED won't draw any current.

```
// LGT8F328P sleep test DPS2

#include <avr\sleep.h>
                                   // Sleep mode 111 binary
#define SLEEP_MODE_DPS2 SLEEP_MODE_EXT_STANDBY

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN,HIGH);
  delay(8000);                     //wait 8 sec before sleep
  digitalWrite(LED_BUILTIN,LOW);
  delay(500);                      //wait more 1/2 sec before sleep

  cli();
  IOCWK = 0;                       // port D level change wake up. 0 = No wake up when level changes
  set_sleep_mode(SLEEP_MODE_DPS2);
  sleep_enable(); 
  DPS2R = 1<<DPS2E;                // DPS2 enable (bit 3), LPRC timer disable (bit 2)
  
  sleep_cpu();
}

void loop() {}
```

sullivanzheng, 2023-02-07:
> @alxgarza
> 
> > Has anyone achieved the 1µA in standby mode?
> 
> I tested and reached current lower than 2 µA in deep-sleep mode. (Measured with a 3 and 1/2 digits METEX multimeter.) I was used the SSOP20 Mini style board for testing because only three components had to be removed. The voltage regulator, and 2 LED's resistors. (The LQFP32 Mini style board can be just as good.) At the first test I didn't remove the builtin LED's resistor so it turned out the pull-up resistor switched on when the deep-sleep mode started. After this I removed this resistor and connected an external LED with a 10k resistor between D13 Arduino pin and VCC, because when pull-up resistors switching on this LED won't draw any current.
> 
> ```
> // LGT8F328P sleep test DPS2
> 
> #include <avr\sleep.h>
>                                    // Sleep mode 111 binary
> #define SLEEP_MODE_DPS2 SLEEP_MODE_EXT_STANDBY
> 
> void setup() {
>   pinMode(LED_BUILTIN, OUTPUT);
>   digitalWrite(LED_BUILTIN,HIGH);
>   delay(8000);                     //wait 8 sec for current measure before sleep
>   digitalWrite(LED_BUILTIN,LOW);
>   delay(500);                      //wait more 1/2 sec before sleep
> 
>   cli();
>   IOCWK = 0;                       // port D level change wake up. 0 = No wake up when level changes
>   set_sleep_mode(SLEEP_MODE_DPS2);
>   sleep_enable(); 
>   DPS2R = 1<<DPS2E;                // DPS2 enable (bit 3), LPRC timer disable (bit 2)
>   
>   sleep_cpu();
> }
> 
> void loop() {}
> ```

I replicated your code and you are a genius. I got **0.67uA (670nA)!**

sullivanzheng, 2023-02-07:
@alxgarza you are more than welcome to check other discussions related to power consumption
[More power consumption measurements on DPS2 mode - with or without periodic wake up ](https://github.com/dbuezas/lgt8fx/issues/155#issuecomment-1421186331)

alxgarza, 2023-02-07:
> @alxgarza you are more than welcome to check other [see bottom] discussions related to power consumption. I think it is quite easy to reach proper power saving to your needs:
> 
> ~16uA: DPS0/1 sleep with SRAM retention and with WDT for wake up. ~6-7 uA: DPS0/1 sleep with SRAM retention but w/o WDT. ~5 uA: DPS2 sleep w/o SRAM retention but with LPRC for wake up. ~0.6uA(600nA): DPS2 sleep w/o SRAM retention, w/o LPRC for wake up, only setting PortD as wake up pins.
> 
> Those are measured with Vcc = 3.3V. If you are willing to take the privilege that lgt8f328p can run at full speed at Vcc=1.8V, then these current can be significantly reduced.
> 
> ~1.1 uA: DPS2 sleep w/o SRAM retention but with LPRC for wake up. ~0.2uA(220nA): DPS2 sleep w/o SRAM retention, w/o LPRC for wake up, only setting PortD as wake up pins.
> 
> [More power consumption measurements on DPS2 mode - with or without periodic wake up](https://github.com/dbuezas/lgt8fx/issues/155#issuecomment-1421186331)

Has anyone tried to put it to sleep with a 32.768LHz crystal?  Waking up every second.

sullivanzheng, 2023-02-08:
No I haven't try that since running a crystal usually increases power consumption, unless you would like to sacrifice some power for accurate timing.

LaZsolt, 2023-02-08:
One of the differences between ATmega and LGT8F328P is how to handle of the internal analog devices during sleep.
LGT8F328P in the most of the sleep modes not disable these analog devices except DPS2 sleep mode.
These devices may draw some current if they were in use before sleep: ADC, two Analog Comparators, Low Voltage Detector (LVD is equivalent of BOD), DAC, Internal Voltage Reference, Differential Amplifier, Voltage Division circuit

It is also advisable to disable the two internal oscillators in that case when the external 32 kHz crystal used during sleep.

There is also an option to turn off the FLASH during sleep. This should be set (6th bit of MCUCR) before going to sleep similar to the sleep enable bit in SMCR

In the LGT8Fx databook there is no mention of how the SWD port behaves during sleep. So it might be a good idea to disable during sleep.

And there is a completely undocumented LDO. I don't know what amount of current it consumes. I found the disabling code in the PMU library and in this doc [Porting_from_LGT8FX8D_to_LGT8FX8P_v1.0.0.pdf](https://github.com/dbuezas/lgt8fx/blob/master/docs/Porting_from_LGT8FX8D_to_LGT8FX8P_v1.0.0.pdf)

And the Timer 3. It could be disabled by writing the appropriate bit of PRR1 register.

jayzakk, 2023-02-08:
@LaZsolt , what a nice document you found :)

I feel this side information quite important:
(Deepl translate)

In DPS2 mode, all IOs have a weak internal pull-up resistor (~20K~40K) automatically turned on, which needs to be taken into account when designing the system!

sullivanzheng, 2023-02-09:
> And there is a completely undocumented LDO. I don't know what amount of current it consumes. I found the disabling code in the PMU library and in this doc [Porting_from_LGT8FX8D_to_LGT8FX8P_v1.0.0.pdf](https://github.com/dbuezas/lgt8fx/blob/master/docs/Porting_from_LGT8FX8D_to_LGT8FX8P_v1.0.0.pdf)
>

I found that LDO shutdown code also from PMU lib and that doc you mentioned. And I gave it a test:
Test conditions:
Clean circuit. No LDO, LED, No pin attached to anything except for VCC pin and GND pin connected to joulescope and RST pin pulled up.
Sleep mode: DPS0 or DPS1 (doesn't matter)
WDT: disabled
```
DDRB = 0; DDRC = 0; DDRD = 0; PORTB = 0xff; PORTC = 0xff; PORTD = 0xff; //Stop floating pins
ADCSRA = 0; //I always need to turn off ADC explicitly to get good power saving even if I didn't turn it on.
LDOCR = 0x80;
//LDOCR bits: WCE  na  na  na  PDEN VSEL2 VSEL1 VSEL0
LDOCR = 0b010;  //0b0010 = 0b0011=0b0111=0b0001 7uA, 0b1xxx 42uA
```

Therefore,  VSEL2 VSEL1 VSEL0 doesn't matter. I suspect these bits might be LDO trimming register or output voltage register.
However, if you disable PDEN (I guess it is **Power Down ENable register**) the LDO will be disabled in power down mode (DPS0 and DPS1)

LaZsolt, 2023-03-17:
@sullivanzheng 

> `LDOCR = 0b010;  //0b0010 = 0b0011=0b0111=0b0001 7uA`

This is pretty low consumption with SRAM retention. When you measured this value was the FLASH power and SWD interface shut down?

The FLASH power can be disabled during sleep by the FPDEN bit of MCUCR.
SWD interface can be disabled by the SWDD bit of MCUSR.

<br><br>

I think LDO (which can be disabled by PDEN bit of LDOCR register) is powering some part of the analog modules
such as **I**nternal **V**oltage **Ref**erence. So when LVD module (ATmega BOD equvalent) is enabled during sleep to detect low VCC then LDO must be not disabled.
I will do some test to measure it. The output of the internal reference voltage will be put on AVREF pin if ADTM bit of ADCSRC is set. (测试模式，从 AVREF 端口上输出内部参考电压)

LaZsolt, 2023-03-17:
> I think LDO (which can be disabled by PDEN bit of LDOCR register) is powering some part of the analog modules
such as Internal Voltage Reference. 
I will do some test to measure it. The output of the internal reference voltage will be put on AVREF pin if ADTM bit of ADCSRC is set. 

Not true. :disappointed: 

- To place out on a pin the voltage of IVREF with setting the ADTM bit of ADCSR isn't working. But I could measure it on the DAC output pin.
- When LDO was powered down the IVREF voltage does not change.

LaZsolt, 2023-03-28:
> ## How to use sleep modes on LGT8F328P ?

## The anwer is the LGT LowPower library.
Just uploaded: https://github.com/LaZsolt/lgt8fx/tree/master/lgt8f/libraries/lgt_lowpower

It's under constuction, not tested, but I look forward to your questions.

dbuezas, 2023-03-28:
@LaZsolt when you think it is good enough we can add it to the core libraries or at least link it in the readme and add an example to the core, wdyt?

LaZsolt, 2023-03-28:
I need one or two weeks for publish.
The readme has some garbage in the middle and I would like to make some tests and more make examples (eg. to present how to turn off analog modules).

dwillmore, 2023-03-28:
Do you have examples of all the chips/cores that you need?

LaZsolt, 2023-04-04:
PMU.h : lgt_LowPower.h
1:0

PMU is working, but my adapted LowPower not. The disassembed code is lokking perfect, but somehow not.

LaZsolt, 2023-04-16:
The **lgt_LowPower** library started working finally. The difficulties were caused by faulty WDT operation.

 Today I tested only the Idle sleep mode and created the first 2 examples.
https://github.com/LaZsolt/lgt8fx/tree/master/lgt8f/libraries/lgt_lowpower/examples

I'm already very tired. I will continue the tests in the next few days.

LaZsolt, 2023-04-19:
Today I created and tested one more example. I have written many comments in the source. Aren't these many comments annoying?
https://github.com/LaZsolt/lgt8fx/tree/master/lgt8f/libraries/lgt_lowpower/examples/adcNoiseReduction

dbuezas, 2023-04-19:
For the example your comments are super useful!

dwillmore, 2023-04-19:
> Today I created and tested one more example. I have written many comments in the source. Aren't these many comments annoying? https://github.com/LaZsolt/lgt8fx/tree/master/lgt8f/libraries/lgt_lowpower/examples/adcNoiseReduction

No, absolutely not.  They are very helpful and educational.  Many times I've read my own code and wondered what the heck I was thinking.  To have this insight into the thoughts of someone else is invaluable.  Please be as verbose as you wish to be!  Please don't feel that your comments are in any way a negative.

LaZsolt, 2023-04-20:
Thank you all for your feedback. :slightly_smiling_face:

It used to happen that there were twice as many comments as code in my sources. At the end the code was almost unreadable because lost in remarks.

LaZsolt, 2023-04-25:
Adapting the Low-Power library to LGT8F328 is much more complicated than I thought. I've been doing it for 3 months now, but there are still surprises.
I tested wake-up capabilities of the external interrupt pins yesterday, but not worked. After reading the databook, it turned out that the external interrupt flag bit will not set while interrupts are disabled with `cli`. Interrupts must be disabled with `cli` due WDT error. This error causes the MCU to reboot when the sleep time expires if the interrupts not disabled with `cli`.

So when the sleep period is set to `SLEEP_FOREVER` the external interrupt pins could work if interrupts reenabled before sleep instruction... Well, then I started rewriting the affected parts of the source code. again

LaZsolt, 2023-05-21:
This issue is solved with this PR: https://github.com/dbuezas/lgt8fx/pull/274
More description, how to use in the REAME: https://github.com/dbuezas/lgt8fx/tree/master/lgt8f/libraries/lgt_LowPower
Related release: https://github.com/dbuezas/lgt8fx/releases/tag/v2.0.4

closed 2023-05-21 by LaZsolt

osgood232, 2024-06-11:
Hello! - well what an amazing thread, it kind of points to everything I am looking for. I am developing a small battery device that will be unused for long periods of time, hence I am hoping to minimize current draw in when standby mode.

With regards to below - fantastic - thanks for that, and it works (in its current form on my green miniEVB), however I note that its in setup(). Ive tried it in loop() and it does go into sleep, but using an interrupt on D2 it doesnt wake up again, in other words - how do you wake it up??? I also note that when the code gets to those lines (again in my loop() ) it doesnt draw the low current that I am able to replicate when running exactly that code it in setup(). 

I think the issue will be with my understanding of this. Just for reference, I am a professional engineer exposed to electronics and software but not so much with Arduino/C based code (I couldn't call myself a software engineer) - i would still consider myself a hobbiest in this domain - so any advice will be VERY welcome!

Thanks

> @alxgarza
>
>
> I tested and reached current lower than 2 µA in deep-sleep mode. (Measured with a 3 and 1/2 digits METEX multimeter.) I was used the SSOP20 Mini style board for testing because only [...]

osgood232, 2024-06-11:
Well that was amazing, its now working as I would expect but in "standby" mode its pulling just 7uA (as opposed to my previous reading of ~600uA but still can be woken up! - well done and thanks.

Now to get into the fine tuning, there maybe more questions later on...

Trotyl64, 2025-07-21:
With the following code using the PMU library I managed to get around 20uA current draw with a 3.3V LDO(which should have a quiescent current of 2.5uA), earlier I was getting 250uA with a 5V one. Either the MCU uses a lot more current at 5V or the LDO was fake. I have what I needed - sleep for long durations of time.
```
#include <PMU.h>
#include <lgt8fx_power.h>

void setup() {
    DDRB  = 0x00;                 // Set all pins to input
    DDRC  = 0x00;
    DDRD  = 0x00;
    DDRE  = 0x00;
    PORTB = 0xFF;                 // Set all input with internal pull up resistors
    PORTC = 0xFF;
    PORTD = 0xFF;
    PORTE = 0xFF;
    pinMode(LED_BUILTIN, OUTPUT); 
    digitalWrite(LED_BUILTIN, LOW);

    digitalWrite(LED_BUILTIN, HIGH);
    delay(4000);
    digitalWrite(LED_BUILTIN, LOW);
}

void loop() {
    digitalWrite(LED_BUILTIN, LOW);

    power_pci_disable();
    power_timer3_disable();
    power_timer0_disable();
    power_timer1_disable();
    power_timer2_disable();
    power2_ac0_disable();
    power2_ac1_disable();
    power2_spi_disable();
    power2_usart_disable();
    power2_usart0_disable();
    power2_twi_disable();
    power2_dap_disable();
    power2_uDSC_disable();
    power2_swd_disable();
    power2_dac_disable();
    power2_adc_disable();
    power2_ldo_disable();

    //not sure if these are needed with power2_adc_disable()
    ADCSRA &= 0x00;
    ADCSRA |=  (1 << ADIF);
    //sleep for 4*32 seconds = around 128 seconds
    for (int i = 0; i < 4; i++)
    {
        PMU.sleep(PM_POFFS1, SLEEP_32S);
    }

    power2_ldo_enable();
    power2_dac_enable();
    power2_adc_enable();
    power2_ac0_enable();
    power2_ac1_enable();
    power2_spi_enable();
    power2_usart_enable();
    power2_usart0_enable();
    power2_twi_enable();
    power2_dap_enable();
    power2_uDSC_enable();
    power2_swd_enable();
    power_pci_enable();
    power_timer3_enable();
    power_timer0_enable();
    power_timer1_enable();
    power_timer2_enable();

    //blink just for debug
    digitalWrite(LED_BUILTIN, HIGH);
    delay(500);
    digitalWrite(LED_BUILTIN, LOW);
    delay(500);
    digitalWrite(LED_BUILTIN, HIGH);
    delay(500);
    digitalWrite(LED_BUILTIN, LOW);
    delay(500);
    digitalWrite(LED_BUILTIN, HIGH);
    delay(500);
    DoStuff();
}