dbuezas/lgt8fx #203: Anyone figured out how to read internal Vcc?

ErikApption, 2022-04-05:
I tried many Arduinos samples to read the internal voltage but I am not able to get any reading that makes sense, although the ADC is supposed to have a decent precision. None of the samples I've tried, both to get the Ref voltage or to read the internal Vcc worked. Any suggestions?

jayzakk, 2022-04-05:
The fix in #130 would allow to select 4/5th VCC, which would allow to read that data. But currently the ADMUX can only select the first 8 inputs, not the extended ones. You may be able to merge it for yourself.

LaZsolt, 2022-04-06:
Try to use this program: https://github.com/LaZsolt/Arduino_Vcc
More info in this link: https://github.com/dbuezas/lgt8fx/discussions/123

LaZsolt, 2022-04-06:
One more solution: https://github.com/RafaelReyesCarmona/Vcc

ErikApption, 2022-04-06:
> Try to use this program: https://github.com/LaZsolt/Arduino_Vcc More info in this link: #123

wow thanks for sharing this one. that's exactly what I was looking for

closed 2022-04-06 by ErikApption