dbuezas/lgt8fx #204: Suppoort board with 16MHz crystal

andrey-belokon, 2022-04-21:
Hi,
I buy on ali boards with 16MHz crystal. In arduino IDE I can select only 32MHz as external clock. Please add support for 16MHz external clock.
Thanks, Andrey

LaZsolt, 2022-04-22:
Maybe this will works with 16 or 12 MHz. https://github.com/dbuezas/lgt8fx/blob/fb4f06a097e6a225a20d1b78d7e70eac160a6d5b/lgt8f/boards.txt
 if you replace your own copy.

Here are some pictures of how the menu will look like: https://github.com/dbuezas/lgt8fx/pull/119#issuecomment-1006800753

andrey-belokon, 2022-04-28:
I try to replace boards.txt, but simple blink.ino compiled with error

`c:\program files (x86)\arduino\hardware\tools\avr\avr\include\util\delay.h: In function 'void _delay_ms(double)':
<command-line>:0:8: error: 'build' was not declared in this scope
In file included from C:\Users\Andre\AppData\Local\Arduino15\packages\LGT8fx Boards\hardware\avr\1.0.6\cores\lgt8f/USBAPI.h:27:0,
                 from C:\Users\Andre\AppData\Local\Arduino15\packages\LGT8fx Boards\hardware\avr\1.0.6\cores\lgt8f/Arduino.h:243,
                 from sketch\LGT8F328P_16mhz.ino.cpp:1:
c:\program files (x86)\arduino\hardware\tools\avr\avr\include\util\delay.h:174:19: error: invalid operands of types 'void' and 'double' to binary 'operator/'
  __tmp = ((F_CPU) / 1e3) * __ms;
           ~~~~~~~~^~~~~
c:\program files (x86)\arduino\hardware\tools\avr\avr\include\util\delay.h: In function 'void _delay_us(double)':
<command-line>:0:8: error: 'build' was not declared in this scope
In file included from C:\Users\Andre\AppData\Local\Arduino15\packages\LGT8fx Boards\hardware\avr\1.0.6\cores\lgt8f/USBAPI.h:27:0,
                 from C:\Users\Andre\AppData\Local\Arduino15\packages\LGT8fx Boards\hardware\avr\1.0.6\cores\lgt8f/Arduino.h:243,
                 from sketch\LGT8F328P_16mhz.ino.cpp:1:
c:\program files (x86)\arduino\hardware\tools\avr\avr\include\util\delay.h:263:19: error: invalid operands of types 'void' and 'double' to binary 'operator/'
  __tmp = ((F_CPU) / 1e6) * __us;
           ~~~~~~~~^~~~~
exit status 1
`

LaZsolt, 2022-04-28:
Last time I did not write the good location of `boards.txt`
`C:\Users\ (Your username) \AppData\Local\Arduino15\packages\LGT8fx Boards\hardware\avr\1.0.6\`

If you have placed `boards.txt` this locacion `C:\Program Files (x86)\Arduino\hardware\arduino\avr\`, then you need to retrieve the original file here.

tistructor, 2022-06-08:
The alpha 1.0.7 version works well and the timing is also correct with the baudrate of the serial port.

Version 1.0.6 instead causes the compilation error.
Con la versione modificata boards.txt

dwillmore, 2023-01-12:
Can you verify that the release 1.0.7 corrects this issue as well and close this issue if so?  Thank you.

LaZsolt, 2023-01-31:
> Support board with 16MHz crystal

With release v2.0.0. the support is done.

> Can you verify that the release 1.0.7 corrects this issue as well

@tistructor verified.

So I close this issue.

closed 2023-01-31 by LaZsolt