dbuezas/lgt8fx #210: Nano 328P-LGFP32 V3.1

biblical-text, 2022-06-03:
[image]

I purchased this thinking it was a standard Arduino Nano, and I can actually deploy from the Arduino IDE using the "Arduino Nano" setting. But I can't deploy to it using any of the options from this library. Should I be able to? I tried every menu option and  it always returns the dreaded:

```
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
```

LaZsolt, 2022-06-03:
The upload speed is 115200 baud for this board.
You may select this upload speed if you install the pre release of lgt8fx package:
https://github.com/dbuezas/lgt8fx/discussions/207

You need to select a custom board: LGT8F328 + 16 MHz crystal
but your board has no external crystal so you need to select one of the internal frequencies.

[image]

demosspro, 2022-06-08:
> The upload speed is 115200 baud for this board. You may select this upload speed if you install the pre release of lgt8fx 
Hello! Is Daniel here and we  can comit some changes? Do you have some contacts with him? lgt8f328p is good chip and i do some cool things with it. I would be grateful if we could develop it further with you!

dwillmore, 2023-01-12:
Does this still persist in 1.0.7 release?

dbuezas, 2023-03-30:
Closing as stale

closed 2023-03-30 by dbuezas