dbuezas/lgt8fx #223: 48 pins device  : D32 is wrong

trimarco232, 2022-10-14:
D32 is PORTF, 5  ;  it must be   PORTE, 5
(D37 is PORTF, 5)
... didn't find the reason of the issue ...

LaZsolt, 2022-10-27:
I think I found the reason of the issue.
In file [fastio_digital.h](https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/cores/lgt8f/fastio_digital.h) the port F referenced more than 8 times.
Must be changed in line [38](https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/cores/lgt8f/fastio_digital.h#L38) the DDRF to DDRE.
Must be changed in line [89](https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/cores/lgt8f/fastio_digital.h#L89) the DDRF to DDRE.
Must be changed in line [140](http://github.com/dbuezas/lgt8fx/blob/master/lgt8f/cores/lgt8f/fastio_digital.h#L140) the PORTF to PORTE.
Must be changed in line [191](http://github.com/dbuezas/lgt8fx/blob/master/lgt8f/cores/lgt8f/fastio_digital.h#L191) the PORTF to PORTE.
Must be changed in line [242](http://github.com/dbuezas/lgt8fx/blob/master/lgt8f/cores/lgt8f/fastio_digital.h#L242) the PINF to PINE.
Must be changed in line [293](http://github.com/dbuezas/lgt8fx/blob/master/lgt8f/cores/lgt8f/fastio_digital.h#L293) the PINF to PINE.

LaZsolt, 2022-11-23:
@trimarco232 
Have you tested the modifications to see if they work well?

LaZsolt, 2023-05-09:
I fixed fastio_digital.h : https://github.com/dbuezas/lgt8fx/commit/b2f00c8913a6857ef5486b30997d15616a14e858

trimarco232, 2023-05-09:
Hi,
ok, a great fix for a great board (the purple one) , thanks !
note that fastioMode(25, OUTPUT) , is not enough to make the pin 25 a digital pin : use, for example, pinMode(25, OUTPUT) , before , typically in the setup

closed 2023-05-09 by trimarco232

LaZsolt, 2023-05-09:
@trimarco232 
Thank you for your feedback.