dbuezas/lgt8fx #225: LGT8F328P-nano as isp

orabeidm, 2022-11-24:
vcan i use LGT8F328P-nano as isp for arduino nano?

orabeidm, 2022-11-24:
[image]

jayzakk, 2022-11-24:
One of the first results on Google reveal: https://youtu.be/KMzk5gsDQMc

closed 2022-11-24 by jayzakk

LaZsolt, 2022-11-24:
Maybe this is the answer for you: https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP
Use your LGT8F328P as a programmer and the classic nano3 as a target.

LaZsolt, 2022-11-24:
And one more. Try tu use this: https://github.com/dbuezas/lgt8fx/discussions/207

orabeidm, 2022-11-24:
> 

> One of the first results on Google reveal: https://youtu.be/KMzk5gsDQMc
> 
> Double check the question and answer on the video.

Yes, and in the comments we communicate with the creator of the video)))

orabeidm, 2022-11-24:
> Maybe this is the answer for you: https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP
> Use your LGT8F328P as a programmer and the classic nano3 as a target.

it doesn't work i checked

LaZsolt, 2022-11-24:
> it doesn't work i checked

In v1.0.6 the delayMicroseconds() about 30% shorter than requested. Use the pre-release of v1.07 : #207