dbuezas/lgt8fx #227: Serial Port with clock at 4 MHz and below not working

holgerlembke, 2023-01-08:
Moin.

I use a LGT8F328P-nano style board.

Setup is spectacular code!
```
void setup() {
  Serial.begin(115200);
  delay(100);
  Serial.print(F("\n\nBenjaNano "));
  Serial.print(F(__DATE__));
  Serial.print(" ");
  Serial.print(F(__TIME__));
  Serial.println();

  setupHeartBeat();
}
```
With 1 MHz, 2 MHz and 4 MHz the output is garbage, 8 MHz, 16 MHz and 32 MHz work fine.

I just switch the clock setting (new IDE) and reupload.

LaZsolt, 2023-01-08:
Look at this tool: [WormFood's AVR Baud Rate Calculator Ver. 2.1.1 25](http://wormfood.net/avrbaudcalc.php) , it shows you how the baud rate can work with different system clocks.

LaZsolt, 2023-01-08:
The real serial speed what the hardware does at 2 and 4 MHz is 100kbaud.
It is not a software issue.

holgerlembke, 2023-01-08:
Thanks for the fast response.

I tested 1 MHz with 9600 baud and it works perfectly.

So I assume we have a software serial port that depends on cpu clock, right?

LaZsolt, 2023-01-08:
> So we have USART hardware and the clock is generated from CPU clock.

Yes, you are right. 
Despite of LGT8F328P has a 32MHz internal clock, the USART uses the CPU clock.

closed 2023-01-08 by holgerlembke