dbuezas/lgt8fx #232: 1.0.7 has changed analogRead functionality which differs from normal!  Beware!

dwillmore, 2023-01-12:
1.0.7 was released with a changed default analogRead functionality which differs from earlier 1.0.x versions of the library.  Please be aware of this.  1.0.7 is being pulled from distributioni.  A new 2.0.x branch will be opened as this is an API breaking chagne.  Please look for that version shortly.  We are pondering a release of a 1.0.8 with the analogRead functionality removed--which will allow people who need the old functionality to use existing code without changes.

Expect more information shortly.  Sorry for the confusion.

dbuezas, 2023-01-12:
v1.0.7 is now fully deleted, and v2.0.0 is released.

LaZsolt, 2023-01-12:
@dwillmore 

Do you mean the changed default analog resolution from 12 to 10?
For the backward compatibility with the ATmega328P of the default analogReadResolution changed to 10.

dwillmore, 2023-01-12:
My understanding is that all releases through 1.0.6 used 12 bits A/D and 1.0.7 moved to the Arduino compatable default of 10 bit A/D.  Since this would break existing code expecting to see 0-4094 (?) it was decided to push that change to a 2.0.x release as it's a change in the API.

I'm sure David can provide more details in the morning.

dbuezas, 2023-01-12:
Yes, and it was a good change, our only concern is that it will break code made for v1.0.6. By changing the major version, we're signaling breaking changes from the older version :)

closed 2023-01-20 by dwillmore