dbuezas/lgt8fx #244: java.lang.NullPointerException when upload program to LGT8f328 IC

manhabc, 2023-01-27:
When upload program to LGT IC, I got this error. I'm using IDE 1.8.14, LGT board 2.0 (with 1.07 it's ok to upload any program to LGT IC) win7 64bit
Please help me out of this issue. Many thanks
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingProgrammer(SerialUploader.java:295)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:90)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2055)
	at java.lang.Thread.run(Thread.java:748)

LaZsolt, 2023-01-27:
Strange. On my PC at home, running Windows, the upload works without errors.
Try to update your Arduino IDE because Arduino IDE 1.8.19 is the latest.

On the other hand, when I started Arduino IDE on my PC at work and waited until message "Updates available for some of your boards and libraries", the lgt8f boards disappeaed from menu and not listed as updatable board. This may have been due to the change in the name of the package path from "LGT8fx boards" to "lgt8fx".

LaZsolt, 2023-01-27:
I found a possibile solution with Google. https://forum.electro-smith.com/t/issues-when-uploading-to-arduino/238/4

Quote from the forum:
> "I just fixed the wiki to properly mention using the ‘Upload’ button, and also included a note to not use the ‘Upload Using Programmer’ button."

This could mean you need to change the upload method.

LaZsolt, 2023-01-27:
@hmeijdam 
Thank you for the link of the article. When I updated my the lgt8fx package on my home PC the folder `C:/Users/(your username)/AppData/Local/Arduino15/packages/LGT8fx Boards/hardware/avr` became empty for some reason.
So that was the reason of the error free uplodaing.

@manhabc please let us know if the upload went well after deleting the eralier package folder. `LGT8fx Boards`

manhabc, 2023-01-29:
@LaZsolt 
@hmeijdam 
Thank so much, it work well now. So if someone got error like me, just deleting all the earlier LGT folder in Arduino15

closed 2023-03-26 by dbuezas