dbuezas/lgt8fx #253: NO SSL for json file

Digitcom-srls, 2023-03-21:


dbuezas, 2023-03-21:
What fo you mean? 

It is https

https://raw.githubusercontent.com/dbuezas/lgt8fx/master/package_lgt8fx_index.json

Digitcom-srls, 2023-03-21:
[image]

dwillmore, 2023-03-21:
I'm looking at it right now and the github.com key is signed by digicert and is completely valid.  

Possibly you're at a school or business which is monitoring SSL traffic and they are substituting a different certificate and doing it wrong?  Can you show the information page about the certificate that your browser has been presented?

Digitcom-srls, 2023-03-21:
[image]

Digitcom-srls, 2023-03-21:
I fixed it by downloading the json file with my mobile phone and uploading it to a webserver in localhost.

closed 2023-03-21 by Digitcom-srls

jayzakk, 2023-03-21:
This is a fake certificate from somewhere in your network chain, maybe even some local firewall or antivirus scanner.

GitHub sends a very valid certificate. If not, millions of user would have problems.