dbuezas/lgt8fx #258: Cannot compile

willbelr, 2023-03-30:
Settings
```
Board: LGT8F328
Clock source: Internal
Clock: 32 MHz
Variant: 328P-LQFP32
Arduino as ISP: default (64)
Programmer: AVR-ISP

LGT8FX version: 2.0.1
Arduino IDE: 1:1.8.19-2
OS: Linux 6.2.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 22 Mar 2023 22:52:35 +0000 x86_64 GNU/Linux
```

When trying to compile any code for that board, I get this error:
```
/home/user/.arduino15/packages/lgt8fx/hardware/avr/2.0.1/cores/lgt8f/main.cpp:168:10: error: #elif with no expression
     #elif
          ^
Using library Arduino-IRremote at version 2.2.1 in folder: /home/user/Arduino/libraries/Arduino-IRremote 
exit status 1
Error compiling for board LGT8F328.
```

Hence, I edited main.cpp so line 168 becomes "#elif F_DIC <= 256" instead of just "#elif". Then it compile a little further and end up with that error:

```
fork/exec /home/user/.arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc-ar: permission denied
Error compiling for board LGT8F328.
```

LaZsolt, 2023-03-30:
In the line 168 should only contain #else

But there must be another error somewhere.

willbelr, 2023-03-30:
This fixed the problem: https://github.com/MCUdude/MicroCore/issues/21#issuecomment-513410807
Then I reinstalled LG boards along with "Arduino AVR Boards", changed the elif to else, restarted the IDE and it works!

dbuezas, 2023-03-30:
It must be using a different compiler for some reason. In the meanwhile, I'm releasing v2.0.2 with the `#elif` replaced with an `#else`

dbuezas, 2023-03-30:
v2.0.2 released. It's so convenient now, merge a PR and there's a new version deployed in a minute :) 

I'll link to that solution in a pinned comment

closed 2023-03-30 by dbuezas