dbuezas/lgt8fx #263: The WDT issue.

LaZsolt, 2023-04-06:
As @wollewald mentioned in his article most of the LGT8F328P have Watchdog error. This error causes that the watchdog interrupt not working and jumping to the reset vector instead. So with this error the watchdog timer mainly can only be used for watchdog reset.
 - When WDT set to watchdog reset mode its working as expected.
 - When WDT set to watchdog interrupt mode and when the timer expires the WDIF bit will be set in WDTCSR register and the program will continue to run until interrupts not enabled. After enabling the interrupts the CPU immediately jumps to reset vector, but the WDIF bit remains set.

It is possible to create a code snippet in the init procedure which detecting the WDT error. If the error exist it is detectable trough WDIF bit. If no WDT interrupt error in the chip  the WDIF bit will be cleared by WDT interrupt and no jump to reset vector.
At the end a flag can be set for a user can see if the chip is faulty.

wollewald, 2023-04-07:
Sounds good! Thank you for following up on this topic.

LaZsolt, 2023-04-08:
> a flag can be set for a user can see if the chip is faulty.

I changed my mind. It is not necessary to set a signal, but restore WDT interrupt capability by software.

If this function below inserted into `main.cpp` the WDT interrupt capability will be restored. The only difference is the WDT interrupt + reset function will be longer about 64 ms.
This process not tested on 328D yet. I will test it on 328D today.
```
#if defined(__LGT8FX8P__)
void softRepairedWdtIsr(void) __attribute__ ((naked)) __attribute__ ((used)) __attribute__ ((section (".init0")));
void softRepairedWdtIsr(void)
{
    uint8_t tmpwsr = WDTCSR;
    if (tmpwsr & ( (1 << WDIF))) {
        if (tmpwsr & ( (1 << WDIE))) {
            WDTCSR = tmpwsr & ~ (1 << WDIE);
            WDTCSR = tmpwsr & ~((1 << WDIE) | (1 << WDIF));
            // WDT interrupt
            __asm__ __volatile__ ("jmp __vector_6 \n");
        }
        // WDT interrupt then system reset.
        __asm__ __volatile__ ("call __vector_6 \n");
        WDTCSR = (1 << WDE) | (1 << WDCE);
        WDTCSR = (1 << WDE);
        fastioToggle(D13);
        while(1);
    }
}
#endif
```

LaZsolt, 2023-04-08:
> If this function below inserted into main.cpp the WDT interrupt capability will be restored.

It is better to include it in `WDT.h` because there will be no wasted FLASH space when the Watchdog timer not in use.

LaZsolt, 2023-04-08:
Ohhh. Must save and restore SREG before jumping to WDT ISR ( __vector_6 ).

LaZsolt, 2023-04-08:
And some registers. But because I don't know which registers the compiler put there, all of the above must be written in assembler language.

LaZsolt, 2023-04-09:
Done. I finished the better watchdog timer library. Pls try it: https://github.com/dbuezas/lgt8fx/tree/Better_wdt_lib/lgt8f/libraries/WDT

wollewald, 2023-04-15:
@LaZsolt, you are a hero! I tried the wdt_interrupt sketch and it's working perfect. 

In particular, I was missing the ISR to be able to perform actions before the reset. For example, that could be saving data to the EEPROM which can help to find the reason for the WDT reset. wdt_ienable() does not trigger a reset, but if wdt_ienable(WTO_4S) in line 34 is changed to wdt_enable(WTO_4S); I get exactly what I wanted (ISR + reset).

[image]

Great thanks - I will modify this section my blog post.

dbuezas, 2023-04-15:
That's great!
@LaZsolt would you open a PR so it gets auto deployed after we merge?

LaZsolt, 2023-04-15:
Okay, but a bit later.
I would like to finish the LowPower library and after that I will open PR.