dbuezas/lgt8fx #264: lgt8fx - Boards Manager Menu error / quirk - for Internal 16MHz / 32MHz

jlsilicon, 2023-04-15:
I am working with the Lgt8f32x -32 / -48 chips.
The chips have on board 16MHz crystals, but I want to use them at 32MHz.

So, I set :
    Board = Lgt8F328 , Internal 32MHz
- but this runs at 16MHz - delay(1000) 1 second, baud 4800=4800 on terminal

But if I set :
    Board = Lgt8F328 , Internal 16MHz
- but this runs at 32MHz - delay(1000) 0.5 second, baud 4800=9600 on terminal

To verify, I set:
    Board = Lgt8F328 , External 16MHz
- this runs at 16MHz at 16MHz again- delay(1000) 1 second, baud 4800=4800 on terminal

So it seems that the Int 16MHz / Int 32MHz SEttings are backwards

LaZsolt, 2023-04-15:
Please attach a picture of your board setup like this:

[image]

dbuezas, 2023-04-15:
Actually, why are there "internal" options for the board with external clock? It should be enough to select the normal board (which uses the internal one).
Can you try if that works fine? If so, I'll delete those variants

LaZsolt, 2023-04-15:
I tested with this sketch all the combinations of frequencies, but I can't find any error.
```
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(4800);
  Serial.println("Running...");
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
}
```

vklimk, 2023-04-15:
> Actually, why are there "internal" options for the board with external clock? It should be enough to select the normal board (which uses the internal one). Can you try if that works fine? If so, I'll delete those variants

Please don't delete them. It is very useful sometimes to have ability to switch frequency from default one.

jlsilicon, 2023-04-15:
My External Clk = 16MHz 
But I want to use it at 32MHz (Internal Clock)

It works at 32MHz when I set it as settings Clock = Internal 16MHz
But when I set settings Clock = Internal 32MHz - it works at 16MHz 
???

jlsilicon, 2023-04-15:
Reason for this testing - is that I am testing higher prescale rates and higher SPS rates.

I found that loweromg prescale to 3 instead of 5 (512Khz at Cpu=16MHz to 2MHz at Cpu=32MHz) - seems to still give accurate Analog reads :

ADCSRA |= 4 ;    // - seems to be good at i=4+ (pre=16+) => 2MHz @ cpu=32MHz

jlsilicon, 2023-04-15:
[image]

dbuezas, 2023-04-15:
Why have a dedicated "Clock Source Internal" inside the board "LGT8F328 + 16 MHz crystal"? Isn't this the same as just picking the "LGT8F328" board?

I don't mean to remove the  "LGT8F328 + 16 MHz crystal", just the option within it to use the internal clock (which should be the same as the plain "LGT8f328" board.

jlsilicon, 2023-04-15:
It can stay.

But the Int32MHz / Int16MHz problem is still there - with either Option: LGT8F328 or LGT8F328 +16MHzCrystal.

dbuezas, 2023-04-15:
I cannot reproduce any of this ( am using a LQFP48) and neither can @LaZsolt. 
With internal or external clock selections, all is working as intended
[image]

@vklimk The menus are duplicated: `internal clocks` on "LGT8F328 + 16 MHz crystal" are the same as picking "LGT8F328" as a board. Is there anything else that I am missing? This duplication makes the boards.txt longer than necessary and harder to update so I intend to delete unless there is something I'm not seeing :)

closed 2023-04-15 by dbuezas

LaZsolt, 2023-04-15:
I do not rule out that this is another defective chip series.
If the clock prescaler bit 0 stuck to zero, the delay(1000) will be 0.5 s long when internal 16 or  4 or  1 MHz selected.

jlsilicon, 2023-04-15:
Its not stuck - the baudrate for Term changes with the Clk=##MHz chosen

LaZsolt, 2023-04-15:
platform.txt line 54

jlsilicon, 2023-04-15:
Thats not the file.
I want the file with the calculations with #ifdefs and F_CLK

LaZsolt, 2023-04-15:
main.cpp https://github.com/dbuezas/lgt8fx/blob/e844e1c88a5d6dbef352ca13e38aacf561450f5d/lgt8f/cores/lgt8f/main.cpp#L146-L172

jlsilicon, 2023-04-15:
Ahh cores thanks, I remembered from somewhere.

LaZsolt, 2023-04-15:
Can you test the delay(1000) and serial speed at **internal 4 MHz** (divider = 3) and at **internal 1 MHz** (divider = 5) and **external 8 MHz** (divider = 1)?

jlsilicon, 2023-04-15:
Hold on ...

dbuezas, 2023-04-15:
Yes, that just changes the divider (divisor = crystal/selected_clock).

LaZsolt, 2023-04-15:
> -- Wow, Looks like the Divisor is backwards

Very strage. It means the divisor does not work at all and the CPU always running at 32 MHz internal or 16 Mhz external.

jlsilicon, 2023-04-15:
Bummer

jlsilicon, 2023-04-15:
Well, at least I can purchase Arduino MicroPro clone and 48pin at $1.50

jlsilicon, 2023-04-15:
Hold on ... I am going to try loop cmd timing tests ...

dbuezas, 2023-04-15:
is it the one on the far right of these?
[image]

I'm testing on that one (LQFP48) and clock selection works just fine

jlsilicon, 2023-04-15:
for(i=0; i<10000000; i++) { c+=i; }
Int 32MHz :
Int 16MHz : => 4sec
Ext 16MHz :
Int 8MHz :
Int 1MHz : => 4sec

jlsilicon, 2023-04-15:
This did not change it either :

`
int  R;
//  if(mode == INT_OSC) {
    // prescaler settings
    CLKPR = 0x80;
    CLKPR = 0x00; 
//    CLKPR = 0x01; 
    // switch to internal crystal
    R = PMCR & 0x9f;
    PMCR = 0x80;
    PMCR = R;
    // disable external crystal
    R = PMCR & 0xfb;
    PMCR = 0x80;
    PMCR = R;
//  }

    CLKPR = 0x80;
      CLKPR = 0x00;
//      CLKPR = 0x01;
//      CLKPR = 0x02;
//      CLKPR = 0x03;
//      CLKPR = 0x04;
//      CLKPR = 0x05;
//      CLKPR = 0x06;
//      CLKPR = 0x07;
//      CLKPR = 0x08;

long  i , c=0 ;
for(i=0; i<=1000; i++)
{
  c+= i;
}
`

jlsilicon, 2023-04-15:
Can you test your -48 chip with this : 

`
  // This runs MCU at FULL SPEED (32MHz) . See comment below.
  clock_prescale_set(clock_div_1);
  // RCMCAL = 255;  // for overclock to 42MHz
 
  // Wait for serial to initialise
//  Serial.begin(4800);  // - term at 9600/4800? if Xtal=32MHz/16MHz ? 
  Serial.begin(9600);  // - term at 9600/4800? if Xtal=32MHz/16MHz ? 
//  Serial.begin(115200);  // - term at 230400/115200? if Xtal=32MHz/16MHz ? 
  while ( ! Serial ) 
  {} 
  Serial.println("Booting: ... please wait ... ");

  pinMode(13, OUTPUT);
  digitalWrite(13, HIGH);

long  i , c=0;
Serial.println("START");

// Clk=Int32MHz => 4 sec 
// Clk=Ext16MHz =>  sec 
// Clk=Int16MHz => 4 sec 
// Clk=Int8MHz  =>  sec 
// Clk=Int4MHz  =>  sec 
// Clk=Int2MHz  =>  sec 
// Clk=Int1MHz  => 4 sec 
while(1)
{
  for(i=0; i<=10000000; i++)
  {
    c += i ;
  }

  digitalWrite(13, LOW);
Serial.print("LOW ");
Serial.println(c);

  for(i=0; i<=10000000; i++)
  {
    c += i ;
  }

  digitalWrite(13, HIGH);
Serial.print("HIGH ");
Serial.println(c);
}
}
`

jlsilicon, 2023-04-15:
Is this giving me the 32MHz ? : 
 
clock_prescale_set(clock_div_1);