dbuezas/lgt8fx #265: Lgt8fx - ADC analogread() only gives bac 10bit - need to load manually to get 12bit value

jlsilicon, 2023-04-15:
I used analogread() for Lgt8fx , but it only gave me vbalues from 0 to 1023 (10bits).
I needed to manually read ADCH + ADCL to get the values from 0 to 4095 (12bits)

See below : 

    // - wait for ADC Ready 
    // - tests wrong - only 10 bits , input of potentiometer varies between 0 .. 1023
    A_R = analogRead( A0 );  

    // - get true 12bit value - tests correctly - returns 12 bits , input of potentiometer varies between 0 .. 4095
    A_R = ( (ADCH << 8) | (ADCL << 0) );

jayzakk, 2023-04-15:
10 bits is the Arduino default for compatibility reasons.
If you need better (and know your board supports it) you can change with analogReadResolution().

This works as intended, so closing.

Regards,
Oliver

closed 2023-04-15 by jayzakk

jlsilicon, 2023-04-15:
oh well