dbuezas/lgt8fx #269: FastIO will not accept defined constants

vkvark, 2023-04-30:
Very strange, if I try to compile this sketch:

```
#define led 10

void setup() {
fastioMode(led, OUTPUT);
}

void loop() {
}
```
Arduino IDE will not compile it, giving me the error below:

```
In file included from C:\Users\kvark\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.3\cores\lgt8f/Arduino.h:629:0,
                 from sketch\sketch_apr30a.ino.cpp:1:
C:\Users\kvark\Downloads\sketch_apr30a\sketch_apr30a.ino: In function 'void setup()':
C:\Users\kvark\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.3\cores\lgt8f/fastio_digital.h:414:43: error: '_FIO_PORT_led' was not declared in this scope
  if((dir & INPUT_PULLUP) == INPUT_PULLUP) _FIO_PORT_##pin |= (1 << _FIO_BIT_##pin);\
                                           ^
C:\Users\kvark\Downloads\sketch_apr30a\sketch_apr30a.ino:18:1: note: in expansion of macro 'fastioMode'
 fastioMode(led, OUTPUT);
 ^~~~~~~~~~
C:\Users\kvark\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.3\cores\lgt8f/fastio_digital.h:414:43: note: suggested alternative: '_FIO_PORT_D0'
  if((dir & INPUT_PULLUP) == INPUT_PULLUP) _FIO_PORT_##pin |= (1 << _FIO_BIT_##pin);\
                                           ^
C:\Users\kvark\Downloads\sketch_apr30a\sketch_apr30a.ino:18:1: note: in expansion of macro 'fastioMode'
 fastioMode(led, OUTPUT);
 ^~~~~~~~~~
C:\Users\kvark\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.3\cores\lgt8f/fastio_digital.h:414:68: error: '_FIO_BIT_led' was not declared in this scope
  if((dir & INPUT_PULLUP) == INPUT_PULLUP) _FIO_PORT_##pin |= (1 << _FIO_BIT_##pin);\
                                                                    ^
C:\Users\kvark\Downloads\sketch_apr30a\sketch_apr30a.ino:18:1: note: in expansion of macro 'fastioMode'
 fastioMode(led, OUTPUT);
 ^~~~~~~~~~
C:\Users\kvark\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.3\cores\lgt8f/fastio_digital.h:414:68: note: suggested alternative: '_FIO_BIT_D10'
  if((dir & INPUT_PULLUP) == INPUT_PULLUP) _FIO_PORT_##pin |= (1 << _FIO_BIT_##pin);\
                                                                    ^
C:\Users\kvark\Downloads\sketch_apr30a\sketch_apr30a.ino:18:1: note: in expansion of macro 'fastioMode'
 fastioMode(led, OUTPUT);
 ^~~~~~~~~~
C:\Users\kvark\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.3\cores\lgt8f/fastio_digital.h:415:23: error: '_FIO_DDR_led' was not declared in this scope
     if(dir == OUTPUT) _FIO_DDR_##pin |= (1 << _FIO_BIT_##pin);\
                       ^
C:\Users\kvark\Downloads\sketch_apr30a\sketch_apr30a.ino:18:1: note: in expansion of macro 'fastioMode'
 fastioMode(led, OUTPUT);
 ^~~~~~~~~~
C:\Users\kvark\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.3\cores\lgt8f/fastio_digital.h:415:23: note: suggested alternative: '_FIO_DDR_0'
     if(dir == OUTPUT) _FIO_DDR_##pin |= (1 << _FIO_BIT_##pin);\
                       ^
C:\Users\kvark\Downloads\sketch_apr30a\sketch_apr30a.ino:18:1: note: in expansion of macro 'fastioMode'
 fastioMode(led, OUTPUT);
 ^~~~~~~~~~
C:\Users\kvark\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.3\cores\lgt8f/fastio_digital.h:415:47: error: '_FIO_BIT_led' was not declared in this scope
     if(dir == OUTPUT) _FIO_DDR_##pin |= (1 << _FIO_BIT_##pin);\
                                               ^
C:\Users\kvark\Downloads\sketch_apr30a\sketch_apr30a.ino:18:1: note: in expansion of macro 'fastioMode'
 fastioMode(led, OUTPUT);
 ^~~~~~~~~~
C:\Users\kvark\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.3\cores\lgt8f/fastio_digital.h:415:47: note: suggested alternative: '_FIO_BIT_D10'
     if(dir == OUTPUT) _FIO_DDR_##pin |= (1 << _FIO_BIT_##pin);\
                                               ^
C:\Users\kvark\Downloads\sketch_apr30a\sketch_apr30a.ino:18:1: note: in expansion of macro 'fastioMode'
 fastioMode(led, OUTPUT);
 ^~~~~~~~~~
C:\Users\kvark\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.3\cores\lgt8f/fastio_digital.h:416:10: error: '_FIO_DDR_led' was not declared in this scope
     else _FIO_DDR_##pin &= ~(1 << _FIO_BIT_##pin); \
          ^
C:\Users\kvark\Downloads\sketch_apr30a\sketch_apr30a.ino:18:1: note: in expansion of macro 'fastioMode'
 fastioMode(led, OUTPUT);
 ^~~~~~~~~~
C:\Users\kvark\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.3\cores\lgt8f/fastio_digital.h:416:10: note: suggested alternative: '_FIO_DDR_0'
     else _FIO_DDR_##pin &= ~(1 << _FIO_BIT_##pin); \
          ^
C:\Users\kvark\Downloads\sketch_apr30a\sketch_apr30a.ino:18:1: note: in expansion of macro 'fastioMode'
 fastioMode(led, OUTPUT);
 ^~~~~~~~~~
C:\Users\kvark\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.3\cores\lgt8f/fastio_digital.h:416:35: error: '_FIO_BIT_led' was not declared in this scope
     else _FIO_DDR_##pin &= ~(1 << _FIO_BIT_##pin); \
                                   ^
C:\Users\kvark\Downloads\sketch_apr30a\sketch_apr30a.ino:18:1: note: in expansion of macro 'fastioMode'
 fastioMode(led, OUTPUT);
 ^~~~~~~~~~
C:\Users\kvark\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.3\cores\lgt8f/fastio_digital.h:416:35: note: suggested alternative: '_FIO_BIT_D10'
     else _FIO_DDR_##pin &= ~(1 << _FIO_BIT_##pin); \
                                   ^
C:\Users\kvark\Downloads\sketch_apr30a\sketch_apr30a.ino:18:1: note: in expansion of macro 'fastioMode'
 fastioMode(led, OUTPUT);
 ^~~~~~~~~~
exit status 1
```

LaZsolt, 2023-04-30:
Yes, FastIO can not accept defined constants or variables. It is not an error. FastIO not designed for such a type of source.

hmeijdam, 2023-04-30:
taken from [here](https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/libraries/lgt328p/examples/lgt8fx8p_fastio/lgt8fx8p_fastio.ino)

// =======================================================
// Larduino HSP for LGT8FX8D/P series
// Fast.IO demostration
// =======================================================
// Important note:
//  fast_io interface is implemented by macro definition
//  Compare to arduino's digital i/o interface, fast_io 
//  is small and fast. 
//  But fast_io has some limitations:
//    1. only support digital mode i/o operation;
**_//    2. fastio DO NOT support avriable as arguments!_**
//       e.g, you can not use fastio as following:
//          int led = 10;
//          fastioMode(led, OUTPUT);
//       but you can use it like this:
//          #define led 10
//          fastioMode(led, OUTPUT);
//      or  fastioMode(10, OUTPUT); 
//      or  fastioMode(D10, OUTPUT);
// ========================================================

vkvark, 2023-04-30:
Then you should change the comment:

// but you can use it like this:
// #define led 10
// fastioMode(led, OUTPUT);
// or fastioMode(10, OUTPUT);
// or fastioMode(D10, OUTPUT);

Because this code will not work.

trimarco232, 2024-09-23:
Hi, still the same thing here

marikyukyu, 2026-01-20:
The current implementation for fastiomode uses a macro and it doesn't accept another macro or const as an argument, only the pin number somehow..

We could probably take inspiration from DxCore's because their fastio works with macros and constants too.