dbuezas/lgt8fx #287: Internal clock speeds less than 32MHz no longer available since version 2.0.3.

VE5BNC, 2023-06-22:
Hi

Since the menus were reorganized in version 2.0.3 the only internal clock speed available is 32 MHz.  My testing showed that the internal clock speeds as low as 8 MHz worked correctly in version 2.0.2 (I didn't test any lower than 8 MHz).  Some of my projects run at 8 MHz.  Can these lower internal clock speeds be added again?

Thanks

Bruce

LaZsolt, 2023-06-22:
If you choose the 32 Mhz internal RC clock and divider 4, the 8 MHz CPU clock is given. Is that right for you?

[image]

VE5BNC, 2023-06-22:
Hi

Thanks for the quick reply and suggestion.

I didn't realize that was the effect of the clock divider.  I'll test that
and let you know.

Bruce

VE5BNC, 2023-06-22:
Hi

I just tested using the clock divider and it works perfectly.

Thanks

Bruce

closed 2023-06-22 by VE5BNC