dbuezas/lgt8fx #295: LGT8F328P program size

Schawen, 2023-07-31:
I'm working on a private LoRa project using LGT8F328P (miniEVB) and RFM95W modules.

Compiling and uploading the code worked fine until today when the following problem occurred: 

`Linking .pio\build\LGT8F328P\firmware.elf
Checking size .pio\build\LGT8F328P\firmware.elf
Building .pio\build\LGT8F328P\firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
**Error: TheI'm working on a private LoRa project using LGT8F328P (miniEVB) and RFM95W modules.

Compiling and uploading the code worked fine until today when the following problem occurred: 

`Linking .pio\build\LGT8F328P\firmware.elf
Checking size .pio\build\LGT8F328P\firmware.elf
Building .pio\build\LGT8F328P\firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
**Error: The program size (30224 bytes) is greater than maximum allowed (29696 bytes)**
RAM:   [=======   ]  65.5% (used 1342 bytes from 2048 bytes)
Flash: [==========]  **101.8%** (used 30224 bytes from 29696 bytes)
*** [checkprogsize] Explicit exit, status 1
=========================================== [FAILED] Took 13.91 seconds ===========================================`

But I only changed the pin assignment.

"PlatformIO Home > Project Inspect" results in an other error:

[image]

Any ideas?

dbuezas, 2023-07-31:
You ran out of flash space, maybe some library was updated and it is bulkier?

By the way, 2 kb of flash are reserved for eeprom, so you could claim that. More here: https://github.com/dbuezas/lgt8fx/pull/261#issuecomment-1616703929

dbuezas, 2023-07-31:
The link I posted is to a specific comment, there's what you need to do (basically upload via ISP and ignore warning. Or modify the available flash in boards.txt if the error blocks uploading)

Schawen, 2023-07-31:
Thanks - I'll check this post again ...

But now I have a different problem with LGT8F in pio.
Now I don't even get to that point:

`Processing LGT8F328P (platform: lgt8f; board: LGT8F328P; framework: arduino)
--------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
UnknownPlatform: Unknown development platform 'lgt8f':
  File "C:\Users\MiKo\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 158:
    env.LoadPioPlatform()
  File "C:\Users\MiKo\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\MiKo\.platformio\penv\lib\site-packages\platformio\builder\tools\pioplatform.py", line 66:
    p = env.PioPlatform()
  File "C:\Users\MiKo\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\MiKo\.platformio\penv\lib\site-packages\platformio\builder\tools\pioplatform.py", line 40:
    return _PioPlatform()
  File "C:\Users\MiKo\.platformio\penv\lib\site-packages\platformio\util.py", line 56:
    self.cache[key] = (time.time(), func(*args, **kwargs))
  File "C:\Users\MiKo\.platformio\penv\lib\site-packages\platformio\builder\tools\pioplatform.py", line 36:
    return PlatformFactory.from_env(env["PIOENV"], targets=COMMAND_LINE_TARGETS)
  File "C:\Users\MiKo\.platformio\penv\lib\site-packages\platformio\platform\factory.py", line 100:
    p = cls.new(spec, autoinstall=autoinstall)
  File "C:\Users\MiKo\.platformio\penv\lib\site-packages\platformio\platform\factory.py", line 70:
    raise UnknownPlatform(pkg_or_spec)
============================================ [FAILED] Took 1.63 seconds ============================================`

**platformio.ini**

`
[env:LGT8F328P]
platform = lgt8f
board = LGT8F328P
framework = arduino
board_build.f_cpu = 16000000L
;board_build.f_cpu = 32000000L
;platform_packages = darkautism/framework-lgt8fx@^1.0.6
board_build.clock_source = 1
lib_deps = 
	jgromes/RadioLib@^6.0.0
	adafruit/Adafruit BMP280 Library@^2.6.8
	adafruit/Adafruit Si7021 Library@^1.5.1
	adafruit/Adafruit Unified Sensor@^1.1.9
	;sparkfun/SparkFun Si7021 Humidity and Temperature Sensor@^1.0.5
	;martinl1/BMP280_DEV@^1.0.21
`

whyameye, 2023-10-13:
you should be aware you are using v1.0.6 of this repo from 2020 and the latest is 2.0.7. The latest version is not supported in pio.