dbuezas/lgt8fx #298: WDT clock source problem

gpb01, 2023-08-23:
Hi David,
first of all thank you for the great work.

I need rather short WDT times (_16 ms_), so I tried to modify your WDT.h file to use the HFRC instead of the LFRC by making the following minor corrections:

`

		static void begin() __attribute__((always_inline)) { 
			uint8_t btmp = PMCR | RCMEN;   // 0000 0001 (enable RCMEN)
			        btmp = btmp & ~RCKEN;  // 1111 1101 (disable RCKEN)
				btmp = btmp & ~WCLKS;  // 1110 1111 (select HFRC)
			PMCR = PMCE;  // 1000 0000
			PMCR = btmp;
		}
`

... unfortunately the change seems to have no effect (_I am trying with your example: wdt_interrupt_) and the times are still the same as in LFRC.

What am I doing wrong? What do I need to change to make the WDT work with the HFRC?

Thanks in advance

Guglielmo

_P.S.: Just for your information, I am trying to port the [Arduino_FreeRTOS](https://github.com/feilipu/Arduino_FreeRTOS_Library) library from "feilipu" to LGT8F328 environment and I need the WDT interrupt at 16 ms._

gpb01, 2023-08-23:
@LaZsolt 

Hi, thanks, this is actually the correct way to write what I wanted to do. :-D

Unfortunately, however, **the result does not change** and the WDT continues to operate following the LFRC :-(

Guglielmo

LaZsolt, 2023-08-23:
ATM I don't have time to test but here is an another idea:
```
.
.
.
#include <WDT.h>

LGTWDT Lgtwdt;

void setup() {
    Lgtwdt.begin();
.
.   your code
.
```

gpb01, 2023-08-23:
@LaZsolt 

This seems to be the solution ... 
... by explicitly calling the begin() method, everything seems to work. Thanks for your help !!!

Guglielmo

LaZsolt, 2023-08-23:
Your welcome.

dbuezas, 2023-08-23:
LaZsolt saves the day :)

closed 2023-08-23 by dbuezas