dbuezas/lgt8fx #300: E2PROM

imroend, 2023-09-26:
How to read and write my struct to e2prom?
Not fungstion for eeprom.put() and eeprom.get()

wollewald, 2023-09-26:
It's like on an AVR based Arduino. Try this example:

````
#include <EEPROM.h>

struct myStruct{
    int a;
    float b;
    char c;
};

void setup(){
    Serial.begin(9600);
    myStruct myWriteStruct = {42, 42.42, 'c'};
    unsigned int address = 0;
    EEPROM.put(address, myWriteStruct);

    myStruct myReadStruct = {};
    EEPROM.get(address, myReadStruct);
    Serial.println(myReadStruct.a);
    Serial.println(myReadStruct.b);
    Serial.println(myReadStruct.c);
}

void loop(){}
````
It worked on my LQFP32 MiniEVB

wollewald, 2023-09-26:
If deepl is correct then this is Indonesian and means: "Does it use the EEPROM library for avr?"

The answer is: You can use it like the AVR EEPROM library but it is different "behind the scenes".

Or in Indonesian:

Jawabannya adalah: Anda dapat menggunakannya seperti perpustakaan EEPROM AVR, tetapi berbeda "di balik layar".

I hope this makes sense!

imroend, 2023-09-26:
Saya mendapati niliai kesalahan pada type data int 
16937 //int data
42.42 // float data
C // char data

Jadi untuk int salah , lainnya benar

imroend, 2023-09-26:
[image]

wollewald, 2023-09-26:
Translation is:

_I got an error value on the int data type_
_16937 //int data_
_42.42 // float data_
_C // char data_

_So for int is wrong, others are right_

Don't know what's wrong on your side. I get the correct values. And I can't see the image you tried to upload.
Tidak tahu apa yang salah di pihak Anda. Saya mendapatkan nilai yang benar. Dan saya tidak bisa melihat gambar yang Anda coba unggah.

imroend, 2023-09-26:
Terimakasih , permasalah terselesaikan dengan  menghapus library EEPROM di folder user library/ di my document, mungkin library terbaca di folder tersebut sebelumnya

closed 2023-09-26 by imroend

wollewald, 2023-09-26:
For the benefit of other readers, here's the translation:

Thanks, the problem was solved by deleting the EEPROM library in the user library folder / in my documents, maybe the library was read in that folder before