dbuezas/lgt8fx #312: Won't flash using IDE 2.2.1

bmentink, 2024-02-21:
Hi, I am trying to program LGTBF328P - MiniEVB nano board. I get the following error:

```
avrude error: protocol expects OK byte 0x10 but got 0xfc
avrdude error: unable to read signature data for part ATmega328P, rc=-3
avrdude error: unable to read signature data, rc=-3
Failed uploading: uploading error: exit status 1
```

Any idea's ?

bmentink, 2024-02-21:
Update: I wonder if I got some bad boards. I have 5 boards, 3 program ok, 2 won't.

hmeijdam, 2024-02-21:
You could try to use one of the working ones for attempting to burn the bootloader on one of the defective boards.

bmentink, 2024-02-22:
> You could try to use one of the working ones for attempting to burn the bootloader on one of the defective boards.

Can you show me how to connect the SPI pins? It is not clear from the board which ones connect to MOSI and MISO, CLK etc, seems to have a pin missing. Cheers

hmeijdam, 2024-02-22:
You need [LGTISP](https://github.com/SuperUserNameMan/LGTISP) to burn the bootloader via SWD and SWC.

bmentink, 2024-02-22:
Thanks, will give it a try.

bmentink, 2024-02-22:
I loaded up a good LGT board with the LGTISP.ino sketch ok. However, when I go to program another LGT board connected with SWC,SWD etc it gives error.
```
avrdude error: programmer is out of sync
 ***failed;
avrdude error: programmer is out of sync
 ***failed;
```

hmeijdam, 2024-02-22:
There is a LarduinoISP sketch amongst the examples. Did you compile and load that one?
It also mentions how to wire it up. You need to look at the right column under LGT8FX8P:

[image]

bmentink, 2024-02-22:
You linked the github site, it only shows these files, no examples. Please be specific.
[image]

bmentink, 2024-02-22:
Right, you did not link that initially, now I understand. BUt the question remains, If I use LGT as ESP are the pin connections the same?

hmeijdam, 2024-02-22:
LGT as ISP has these connections. 
[image]
If that does not work, I will dig up my LGT's and hook them up. Has been a few years since I had to bootload one.

bmentink, 2024-02-22:
> LGT as ISP has these connections. [...]

I don't have those names, please give the corresponding D8,D10) etc ..

hmeijdam, 2024-02-22:
Well.... you are posting your question in the Github forum of the core I am using, so I assumed you will have the LarduinoISP as well. But apparently not

Here are the pins 

[image]

bmentink, 2024-02-22:
I have added the core I am using above, is it incorrect?

hmeijdam, 2024-02-22:
That's the one I us
e.

bmentink, 2024-02-22:
> That's the one I use.

So why no examples?

hmeijdam, 2024-02-22:
I honestly have no clue. 
Here is the LarduinoISP sketch

```

// 20 July 2020 David Buezas
// - Bundled and added menu utility
// * When uploading to the programmer,
//   select in the menu: Tools/Arduino as ISP/SERIAL_RX_BUFFER_SIZE to 250)
// * Before using the ISP to program another board, 
//   connect (in the ISP board) the reset pin to gnd via a capacitor,
//   or (at your own risk) short reset to vcc.
// https://github.com/dbuezas/lgt8fx/
//
// author : brother_yan (https://github.com/brother-yan/LGTISP)
//
// LarduinoISP for LGT8FX8P series
// Project fork from
//    - ArduinoISP version 04m3
// Copyright (c) 2008-2011 Randall Bohn
// If you require a license, see 
//     http://www.opensource.org/licenses/bsd-license.php
//
// This sketch turns the Arduino into a AVRISP
// using the following arduino pins:
//
// pin name:    Arduino:          LGT8FX8P:
// slave reset: 10:               PC6/RESET 
// SWD:         12:               PE2/SWD
// SWC:         13:               PE0/SCK
// Make sure to 
// Put an LED (with resistor) on the following pins:
// 9: Heartbeat   - shows the programmer is running
// 8: Error       - Lights up if something goes wrong (use red if that makes sense)
// 7: Programming - In communication with the slave
//
// 23 July 2011 Randall Bohn
// -Address Arduino issue 509 :: Portability of ArduinoISP
// http://code.google.com/p/arduino/issues/detail?id=509
//
// October 2010 by Randall Bohn
// - Write to EEPROM > 256 bytes
// - Better use of LEDs:
// -- Flash LED_PMODE on each flash commit
// -- Flash LED_PMODE while writing EEPROM (both give visual feedback of writing progress)
// - Light LED_ERR whenever we hit a STK_NOSYNC. Turn it off when back in sync.
// - Use pins_arduino.h (should also work on Arduino Mega)
//
// October 2009 by David A. Mellis
// - Added support for the read signature command
// 
// February 2009 by Randall Bohn
// - Added support for writing to EEPROM (what took so long?)
// Windows users should consider WinAVR's avrdude instead of the
// avrdude included with Arduino software.
//
// January 2008 by Randall Bohn
// - Thanks to Amplificar for helping me with the STK500 protocol
// - The AVRISP/STK500 (mk I) protocol is used in the arduino bootloader
// - The SPI functions herein were developed for the AVR910_ARD programmer 
// - More information at http://code.google.com/p/mega-isp

// LarduinoISP for LGTF8FX8P Series
#include "swd_lgt8fx8p.h"

#if SERIAL_RX_BUFFER_SIZE < 250 // 64 bytes的RX缓冲不够大
#error : Please change the macro SERIAL_RX_BUFFER_SIZE to 250 (In the menu: Tools/Arduino as ISP/SERIAL_RX_BUFFER_SIZE)
#endif

#define RESET	10
[... 505 lines cut ...]
  case 'Q': //0x51
    error=0;
    end_pmode();
    empty_reply();
    break;
  case 0x75: //STK_READ_SIGN 'u'
    read_signature();
    break;
    // expecting a command, not CRC_EOP
    // this is how we can get back in sync
  case CRC_EOP:
    error++;
    Serial.print((char) STK_NOSYNC);
    break;

    // anything else we will return STK_UNKNOWN
  default:
    error++;
    if (CRC_EOP == getch()) 
      Serial.print((char)STK_UNKNOWN);
    else
      Serial.print((char)STK_NOSYNC);
  }
}
```

bmentink, 2024-02-22:
THanks, these are the examples I have:
[image]

So just put the above code on the LGT board that has to act as ISP, right?
In that case, what is the  LGTISP.ino I put on from github do then?

hmeijdam, 2024-02-22:
LGTISP.ino may be a newer or older version. I have not used it myself as I usually grab my dedicated SWDICE programmer. 

The sketch I just uploaded here I have used a while back. But I just found my box with LGT's so let me try if it still works.

bmentink, 2024-02-22:
Thanks will wait. Meantime I tried connecting the programming pins at the end of the board to the target board pins, but that did not work either. So Still unsure of pin connections.

hmeijdam, 2024-02-22:
Just tested and works fine

First burn the LarduinoISP.ino to your working Nano

Then hook it up like
```
// pin name:    Programmer NANO         Target LGT8FX8P:
// reset:                   10:            PC6/RESET 
// SWD:                     12:           PE2/SWD
// SWC:                     13:           PE0/SCK
```
Keep AVRISP as programmer and press "burn Bootloader"

[image]

I burned it to a "promini style" LGT board, as my second Nano has no pinheaders yet. But you will select the board you have to get the bootloader that goes with your board.

bmentink, 2024-02-22:
> First burn the LarduinoISP.ino to your working Nano
I keep teling you, I am **not** using an Arduino Nano as ISP, I am using LGT Nano, do I still use pins 10,12,13 on the ISP side ?

hmeijdam, 2024-02-22:
I also used LGT Nano

bmentink, 2024-02-22:
Well I get this using those pinouts:
```
avrdude error: programmer is not responding
avrdude warning: attempt 1 of 10: not in sync: resp=0x00
avrdude warning: ignoring mismatch in unused bits of lock
        (device 0xff != input 0x3f); to prevent this warning set
        unused bits to 1 when writing (double check with datasheet)
avrdude error: programmer is not responding
avrdude warning: attempt 1 of 10: not in sync: resp=0x00
avrdude warning: ignoring mismatch in unused bits of lock
        (device 0xff != input 0x3f); to prevent this warning set
        unused bits to 1 when writing (double check with datasheet)
```

bmentink, 2024-02-22:
I have to go out, will pick this up later. Thanks for the help.

bmentink, 2024-02-22:
I have exactly that!

bmentink, 2024-02-23:
I'm back now if you have any more suggestions

dbuezas, 2024-02-23:
Have you seen the tutorial in this repo?
https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/libraries/LarduinoISP/readme.md

hmeijdam, 2024-02-23:
> Well I get this using those pinouts:
>
> ```
> avrdude error: programmer is not responding
> avrdude warning: attempt 1 of 10: not in sync: resp=0x00
> avrdude warning: ignoring mismatch in unused bits of [...]

The only way how I can get your error is when I do not upload LarduinoISP to the LGT-Nano. 

When I  upload LarduinoISP to it and do not connect a target I get this error instead

```
Arduino: 1.8.19 (Windows 10), Board: "LGT8F328, 250 (to burn ISP), Internal 32MHz, 1, 328P-SSOP20 (e.g. green pseudo pro mini), 57600"

E:\Arduino\arduino-1.8.19\portable\packages\MiniCore\tools\avrdude\7.2-arduino.1/bin/avrdude -CE:\Arduino\arduino-1.8.19\portable\packages\MiniCore\tools\avrdude\7.2-arduino.1/etc/avrdude.conf -v -patmega328p -cstk500v1 -PCOM20 -e -Ulock:w:0x3f:m -Uefuse:w:0xff:m -Uhfuse:w:0xff:m -Ulfuse:w:0xff:m 

<<<< snip  - non informative part >>>>

avrdude stk500_program_enable() error: unable to enter programming mode
avrdude main() error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower the bit clock frequency, e.g. -B 125kHz
        - use -F to override this check
avrdude done.  Thank you.
```
When I connect a target everything works 100%, even without the capacitor or disabled reset pin.

The only other difference I see is that I work on Arduino IDE V1.8.19 as I cannot upgrade to IDE V2 (it has no [portable IDE](https://docs.arduino.cc/software/ide-v1/tutorials/PortableIDE/) anymore)

bmentink, 2024-02-25:
Ok, trying again. I uploaded LarduinoISP to a working LGT sucessfully. Then without connecting target, I try to burn bootloader, I get this error:
`avrdude error: unable to enter programming mode
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower the bit clock frequency, e.g. -B 125kHz
        - use -F to override this check
Failed chip erase: uploading error: exit status 1`
Looks same as yours, except I am on MacOS not Windoze ...

hmeijdam, 2024-02-25:
And what happens if you now connect one of your known working and one of your known not working targets?
The communication between the programmer and your Mac seems to be working.

bmentink, 2024-02-25:
I get the error above. Maybe it's just a dud board?

bmentink, 2024-02-25:
Nope, new working board does the same thing.

hmeijdam, 2024-02-25:
I start to suspect faulty breadboard wires or some HW problem.

LaZsolt, 2024-04-21:
In my opinion the the communication speed is the source if the issue.
The LarduinoISP.ino in line 88 set the speed to 115200,
https://github.com/dbuezas/lgt8fx/blob/307d152b44f4b9f27a07c33ee7e40befbcc6bf80/lgt8f/libraries/LarduinoISP/examples/LarduinoISP/LarduinoISP.ino#L88
but the programmers.txt set the speed to 19200 baud.
https://github.com/dbuezas/lgt8fx/blob/307d152b44f4b9f27a07c33ee7e40befbcc6bf80/lgt8f/programmers.txt#L13

dbuezas, 2024-05-05:
With the old IDE (v1.8.19) the ISP works fine as is. It has happened to me that it failed to work, but the reason ended up being that the boards were fried.

Devilscave, 2024-06-04:
I have to get involved in this topic now. I have exactly the same problem. I have an LGT Nano as target and also an LGT Nano as ISP programmer. I have installed the Larduino and also read the example, which I have also installed on my LGT ISP as prescribed and then connected the ISP to the target as specified.
GND -> GND
5v     -> VCC
D10  -> RESET
D12  -> SWD
D13  -> SWC
I have tried different variants and also set various build rates in code line 88. All without success. It always says Sync does not match or Target does not respond. It is devinitiv a sketch without bootloader on the target, now I have to load a bootloader again but it doesn't like it.
Thats my Error:
```
avrdude: stk500_program_enable(): failed to enter programming mode
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.
Fehler beim Brennen des Bootloaders.
```

Devilscave, 2024-06-04:
OK guys I have found the error, even if I can't really explain it to myself.
Maybe it concerns you too. I have noticed that in all explanations here always a "GREEN" LGT8 Nano with Micro USB is used.
I tried all the time with a "PURPLE" LGT Nano whit USB-C and always failed. 
But I had one of the green ones and used it and did everything exactly as indicated... installed the LarduinoISP.ino and connected it to a MiniEVB as I did before with the purple LGT. But this time it worked immediately in 2-3 seconds.
Then I connected the green LGT Nano to the purple LGT Nano and within 2-3 seconds the bootloader was on. 
And I was able to upload my sketch to the purple LGT via USB as normal.
So in my case it was definitely the purple LGT Nano. In the meantime, I also tried using an Arduino Nano V3 as an ISP, which also only produced sync errors, but others.
I think there should be a strong differentiation between purple and green in the instructions and I think these should also be strictly separated and named individually in further descriptions. Perhaps it is also due to the more precise setting, one difference between the two is that the purple LGT has an external 16MHz crystal while the green one works with the internal 32MHz crystal.
I think this is where you should start. Please try it out for yourself.

dbuezas, 2024-06-05:
The external crystal won't be used unless you specifically configure the board to use it. I made myself a connector with pogo pins and the reset cap and it has been a lot more reliable ever since. This leds me to believe that the ISP is particularly sensitive to bad connection issues (and also prone to them because of the temptation of "quickly" use jumper cables held with the hand)

wollewald, 2024-06-05:
Hi @Devilscave , I am keen to try myself. I understood that the combination of the purple lgt based board and the green does not work. But I am not sure which board you used as programmer and which as target. Can you tell me that detail?

Another point: you said you tried an Arduino Nano V3. This means a conventional one based on an ATmega328P? This is not a surprise that this does not work.

dbuezas, 2024-06-05:
Which purple one, the LQFP48 (larger) or LQFP32?

lalo-uy, 2024-06-05:
> Hi @Devilscave , I am keen to try myself. I understood that the combination of the purple lgt based board and the green does not work. But I am not sure which board you used as programmer and which [...]

Diden't know that Atmega328 don not work.  What about an ESP-12 Uno board?

wollewald, 2024-06-05:
Hi @lalo-uy  , to be more precise: it possible to use a classic Arduino Nano but not if you follow the instructions here: 

https://github.com/dbuezas/lgt8fx/tree/master/lgt8f/libraries/LarduinoISP

In oder to set the Serial RX Buffer to 250 Bytes you can't use the convenient way:

[image]

Since you have to choose the board on which you want to upload the LArduinoISP sketch, you won't see this setting. Therefore, you have to make this setting manually in HardwareSerial.h which you hould find here:

...\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino

If you should choose this way it works (I have tried beacuse I was curious), but I would recommend to spend little money for a second LGT8F328P based board instead.. 

With the ESP-12 Uno board it might work directly if the buffer is >= 250 bytes. Just check if the LArduinoISP sketch compiles. Done in a few seconds!

Devilscave, 2024-06-06:
@Wollewald
> Hallo@Devilscave, ich möchte es unbedingt selbst ausprobieren. Mir ist klar, dass die Kombination aus der violetten, LGT-basierten Platine und der grünen nicht funktioniert. Aber ich bin mir nicht sicher, welche Platine Sie als Programmierer und welche als Ziel verwendet haben. Können Sie mir das genauer erklären?

In my text it says exactly when I used which one. I only tried a purple LGT in my first post and that didn't work.
In my second post where it worked, I used the green programmer as described in the text.
 
> Noch ein Punkt: Du hast gesagt, du hättest es mit einem Arduino Nano V3 versucht. Meinst du damit einen konventionellen auf Basis eines ATmega328P? Es ist keine Überraschung, dass dieser nicht funktioniert.

I figured if it works with a UNO, why not. But I already encountered the error when programming that I can't make the bit settings at all, funnily enough the Nano eats the sketch anyway.

@dbuezas 
> Welches violette Modell, LQFP48 (größer) oder LQFP32?

LQFP32

wollewald, 2024-06-06:
Hi @Devilscave, I can reproduce the issue with the purple LQFP32. It does not work as a programmer, at least when following the standard procedure. I tried burning the bootloader on a green LQFP32 as well as on a purple LQFP32 using the purple LQFP32 as programmer. I also tried a purple LQFP48 as a programmer and that worked perfectly. So, it is a specific problem of the purple LQFP32. 

I don't know what the problem is and how to fix it. My gut feel says it is related to the CH9340C USB-to-TTL Chip on the purple LQFP32 board and how it is wired. But I have no proof for that.

dbuezas, 2024-06-06:
That's very odd. 
[isp.zip](https://github.com/user-attachments/files/15686131/isp.zip)
I don't have a purple LQFP32 to test it my self, would you test this isp code with it?
The only difference is that the pins change:

- SWD    -> PC5 - A5
- RESET -> PC4 - A4
- SWC    -> PC3 - A3

Since the pcb layout is different, I wonder if some pin is either not connected or has some unwanted pulldown or capacitive coupling with the xtal on PB6 and 7 (same port as the original isp code uses).

Regarding the USB to TTL chip, what is your hunch based on? the serial part should only affect pc <-> isp, and that obviously works if serial burning works.

Also, it may be worth trying setting the clock differently, e.g using the external one.

Devilscave, 2024-06-06:
I have to try it again now, because yesterday I must have shot something on one of the purple LGTs, it works to load sketches on it, but runs much too fast and the sketches only make measurements, it almost seems to me as if the bootloader is wrong.

wollewald, 2024-06-07:
Hi @dbuezas , I have tried your sketch and get the same error like before:

[image]

In order to confirm that I have not wired something wrongly, I have done the same with green LQFP32 and it worked. 

Then I found something interesting. I changed back to the purple LQFP32 and uploaded the standard LArduinoISP sketch. Then I connected the purple LQFP32 to a USB-to-TTL Adapter instead of the PC directly:
VCC -> VCC
GND -> GND
RX -> TX
TX -> RX

It looked like this:

[image]

All the rest is set up as one would do using the normal procedure. And it works! I was able to burn the the bootlader and to upload sketches using the purple LQFP32 as a programmer. I really think the problem is the CH9340C Chip. 

There are also purple LQFP32 boards available which other USB-to-TTL Chips than the CH9340C. I have order some from China, so it will take a while until I can test. If anyone reads this and has such a board, it would be great to know if these can be used as programmer. 

@Devilscave, can you check which chip you have on your board?

Devilscave, 2024-06-07:
> @Devilscave, können Sie überprüfen, welchen Chip Sie auf Ihrer Platine haben?

Ich hab auch den CH340C drauf auf allen
Der grüne LGT hat den HT42B534-2

wollewald, 2024-06-07:
@Devilscave , thank you. But I don't have the CH340C, I have the CH**9**340C. It's a small difference in the name, but these are dfferent chips. Sorry, but can you double check? 
_@Devilscave , danke. Aber ich habe nicht den CH340C, ich habe den CH**9**340C. Es ist nur ein kleiner Unterschied im Namen, aber es sind verschiedene Chips. Entschuldigung, aber können Sie das noch einmal überprüfen?_

dbuezas, 2024-06-07:
That's very interesting, i really didn't expect that. Serial communication works normally at any baudrate otherwise?

Übrigens, lass uns auf Englisch bleiben, das hier wird wahrscheinlich von anderen auch gelesen.

Or maybe the reset line is held too long? I think that's controlled by the usb-ttl chip. The capacitor trick could serve to test that.

Devilscave, 2024-06-07:
> @Devilscave, danke. Aber ich habe nicht den CH340C, sondern den CH **9** 340C. Es ist ein kleiner Unterschied im Namen, aber das sind unterschiedliche Chips. Tut mir leid, aber können Sie das noch einmal überprüfen? 

No it's a CH340C

[image]

wollewald, 2024-06-07:
@Devilscave , thank you. Helpful information. Then it's something else on the board. Yesterday I tried to use a purple LQFP**48** board which uses a CH340C hip like yours and that work as a programmer.

@dbuezas , usually I take a 10µF capacitor. I also tried a 100 µF and a 100 nF capacitor and no capacitor. It all didn't work. Or what do you mean with capacitor trick?

dbuezas, 2024-06-07:
Yes I meant a cap at the reset pin. I thought maybe the programmer maybe was reset for too long by dtr.

I burnt my isp without bootloader to avoid accidentally flashing it instead of the target device. I wonder if the purple ones come with a different bootloader?

wollewald, 2024-06-07:
I have just burned a new bootloader on the purple LQFP32 and tried again to use it as programmer. But no success. Maybe I will have a good idea over night.

dbuezas, 2024-06-09:
Could it be that the serial doesn't work well at the specific baudrate used when isp programming?

wollewald, 2024-06-17:
I have some news. I have now received purple LQFP32 boards with a CH340C from two different shops and both work as programmers as they should. I double checked that the purple board with the CH9340C still did not work. Since @Devilscave tried a board with a CH340C chip, which did not the difference between the CH340C and the CH9340C should not be the reason. I have burned on all boards the same bootloader, so this is also not the reason. I have no idea yet why some of these boards do not work. Maybe a bad batch of the PCBs? Here's a photo of one board that is working and another which does not. I can't see obvious differences:

[image]

So, good that this is no general issue, but bad that we don't know why the problematic boards behave like they do and how to identify them (before you buy them!).

Devilscave, 2024-06-18:
@wollewald Maybe you could, since you own both boards and they seem to be identical, simply swap the CH340 chips across. If the error moves, it is the chip, if it stays, it is the board.

wollewald, 2024-06-18:
Hi @Devilscave , good idea. Let me check if get the chip desoldered and transferred without damage.

Devilscave, 2024-06-18:
here the Pinouts from the two ICs
[image]

LaZsolt, 2024-06-19:
I have an idea about the problematic serial converter chip, CH9340C.
The pin 8 low level selecting the dedicated baud rate mode, whatever that means.
So lift up pin 8 from PCB for deselect the dedicated baud rate mode.

[13 more replies]