dbuezas/lgt8fx #319: LGT ISP: cap or 5v short of reset pin unnecessary?

dbuezas, 2024-05-05:
I accidentally discovered that uploading as ISP still works without inhibiting reset on the programmer.
It just takes 2 seconds longer to start (since I left the bootloader in my programmer).

I'm considering to remove the extra step of adding a cap to rst (or the dangerous hack of shorting it to 5v) from the [ISP guide](https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/libraries/LarduinoISP/readme.md) , or at least to mark it as optional.

Any thoughts on this?

LaZsolt, 2024-05-06:
> LGT ISP: cap or 5v short of reset pin unnecessary?

Normally not necessary to connect the reset pin neither in the LGT ISP nor in the device you want to program. I regularly uploaded program via LGT ISP  without connected reset pin. In my opinion, the idea to use the capacitor or pull up the LGT ISP reset pin to 5V was to not have to use a different connection than usual.

Anyway, if SWD, SWC used as I/O port, in the device you want to program, then reset pin must be connected to LGT ISP.

dbuezas, 2024-05-06:
I mean the reset pin of the ISP (not the chip being programmed). When one uploads the programmer will reset unless its reset pin isn't forced high, but it still uploads afterwards.

darkautism, 2025-03-14:
After repeatedly damaging my bootloader, I realized that we don’t need to use the risky method of short-circuiting to prevent reprogramming. As long as there are other Arduino boards available, recovery is always possible. We simply need to use another LGTISP to intentionally corrupt the bootloader.

dbuezas, 2025-03-14:
What do you mean? shorting reset (with a capacitor or just shorting) prevents the serial chip to pull down the reset pin itself.
One thing that could be done is reconfigure the reset pin to act as gpio. That's in the PMX2 register.

> By default PC6 pin is taken as reset, write logic one to this bit to disable external
reset, than PC6 is used as normal I/O

Something like
```cpp
PMX2 |= 0b1
```

But be mindful, you'll have to manually reset at the right time if you want to reflash.

darkautism, 2025-03-14:
Oh, I thought this was related to avoiding re-uploading the sketch. Upon closer inspection, it turns out it’s not. My mistake, please disregard me.

LaZsolt, 2025-03-14:
> **shorting reset (with a capacitor or just shorting)** _prevents the serial chip to pull down the reset pin itself_

**or just not connect that wire (DTR)**