dbuezas/lgt8fx #320: Starting adc conversion before idle sleep

dbuezas, 2024-05-06:
@LaZsolt in the idle example of your low power library, you state that [it is recommended to clear ADC interrupt and start conversion flag before sleep.](https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/libraries/lgt_LowPower/examples/idleWakePeriodic/idleWakePeriodic.ino#L33-L35).

And in a gadget I made that certainly makes a difference. Do you know why that is?

For context, the gadget is a pwm controlled desk fan using a 140mm pc fan from a battery boosted to 18v. I'm using an lgt to read a potentiometer and generate a PWM on timer 1. Poor chip got a lousy job.
To save some power I am running the MCU at 1MHz and reading the pot only 4 times a second (sleeping in idle mode for 256ms each cycle). The idle mode is the only one capable of keeping the PWM running it seems.

```cpp
OCR1A = analogRead(A1) * TOP / 1000;

// It is recommended to clear ADC interrupt and start conversion flag before sleep.
ADCSRA |=  ( 1<<ADIF );
ADCSRA &= ~( 1<<ADSC );

LowPower.idle(SLEEP_256MS, ADC_OFF, TIMER3_OFF, TIMER2_OFF, TIMER1_ON, TIMER0_OFF, SPI_OFF, USART0_OFF, TWI_OFF, PCIC_OFF, FLASHCTL_OFF);
```

dbuezas, 2024-05-06:
Oh, to clear BOTH, not to clear one and start conversion. Got it

closed 2024-05-06 by dbuezas

LaZsolt, 2024-05-06:
Ahhh. My sentence can be misunderstood. I will fix it.

dbuezas, 2024-05-06:
Thank you.
Idle is the only mode capable of outputting pwm, right?
From your lib's docs I thought that extended standby would also be capable of that (since one can leave Timer2 on), but PWM is kept off. Is it correct that timer2 can only be used for interrupts while on this mode?

LaZsolt, 2024-05-06:
> Idle is the only mode capable of outputting pwm, right?

I haven't tested it. 

> Is it correct that timer2 can only be used for interrupts while on this mode?

The documentation is not detailed enough. I found this:

_DPS0 is Same as SAVE mode, supported wake-up source include:
Level change of all pins
Watch dog timer wake-up
TMR2 wake up of Asynchronous mode_

So yes, it can be used to generate interrupts. But only in Asynchronous mode? (With external clock source) I never tested timers interrupts or PWM capability.