dbuezas/lgt8fx #322: SoftwareSerial and Serial cannot be used at the same time? Causing resets

RealTadango, 2024-09-17:
I have this code and the board is resetting every 2 seconds:

#include <SoftwareSerial.h>

SoftwareSerial mySerial(5, 6); // RX, TX

void setup() {
  Serial.begin(57600);
  mySerial.begin(57600);
}

Using just the SoftwareSerial or the Serial works fine. Is it possible to use both?

RealTadango, 2024-09-18:
No that is not the issue. I discovered using the internal 32Mhz clock there are a lot of issues. Using the external 16Mhz crystal everything works just fine.

dbuezas, 2024-09-18:
You must have a defective board, the Software Serial lib was specifically made for this chip and is thoroughly tested

hmeijdam, 2024-09-18:
I am having no issues printing at the same time with HW- and SW-serial.
[image]

SuperUserNameMan, 2024-09-18:
> No that is not the issue. I discovered using the internal 32Mhz clock there are a lot of issues. Using the external 16Mhz crystal everything works just fine.

The internal clock is indeed less reliable and precise than external crystals. Since the internal clock is based on a RC, it might be affected by temperature and some randomness related to their fabrication. 

The lake of precision of the internal RC clock source is know to have an impact on the reliability of high speed serial comm. 

I don't remember the LGT8 documentation, but i know STM32 MCU that relies on internal RC clock source do have a register to "recalibrate" it with a better precision.

RealTadango, 2024-09-18:
It is something very strange. I tested 4 boards at internal 32Mhz with an empty loop and just the setup as in my first post and 2 board have a reset loop (fast 4 flashes every 2 seconds) but the other 2 look fine. When using more code all boards work depending on the code (sometimes the 2 board also keep resetting). When using the external 16Mhz they never reset.... This is not a precision issue since there is nothing send or received and there is just an empty loop. Maybe it is a watchdog issue with the internal 32Mhz oscillator causes by the port setup?

SuperUserNameMan, 2024-09-18:
Are they all from the same seller ?
Do they contain the same version of the bootloader ?

I rememebr there was some fake LGT8 chips that were sold at some point.

Depending on your supplier, it is also possible they are rejected chips.

RealTadango, 2024-09-18:
I bought a lot of 10. I will check the bootloaders.

SuperUserNameMan, 2024-09-18:
> Maybe it is a watchdog issue with the internal 32Mhz oscillator causes by the port setup?

You could try `wdt_disable();` with `#include <avr/wdt.h>` ?

https://avrdudes.github.io/avr-libc/avr-libc-user-manual/group__avr__watchdog.html

RealTadango, 2024-09-18:
> You could try `wdt_disable();` with `#include <avr/wdt.h>` ?
>
> https://avrdudes.github.io/avr-libc/avr-libc-user-manual/group__avr__watchdog.html [...]

Yes that stops the rebooting. So it is a watchdog issue with some chips in a specific situation....