dbuezas/lgt8fx #324: incorrect documentation in lgt8f/libraries/lgt_LowPower/examples/powerDown-Dps2/powerDown-Dps2.ino

whyameye, 2024-10-22:
The documentation in powerDown-Dps2.ino states:
`In DPS2 mode, no need to set up IOs against leakage current because all IOs have a weak internal pull-up resistor which automatically turned on. Thats why the built-in LED on the MiniEVB board will be glow dimly during sleep`

This is not correct. Measuring the IO pins they are clearly floating when in DSP2 mode. I verified this running the example unaltered in several ways:
- connecting any D port pin to GND through a 100K resistor and pin measures 0V when in DSP2
- connecting any D port pin to Vcc through 100K resistor and pin measures Vcc when in DSP2
- with nothing connected to the pin just touching the D2 or D4 pin with a probe from the multimeter sometimes wakes the MCU up from DSP2

The manufacturer's documentation (v1.0.5 English) confirms that the pull-ups will not be on. It states: ` In DPS2 mode, because of core voltage disable and register data lost, interface control status all resume to input status, all IO output drive and pull-up will disable.`

I tested with a MiniEVB board and the built-in LED did not glow dimly.

I can submit a PR if you wish.

dwillmore, 2024-10-22:
Please do.

whyameye, 2024-10-22:
I'm happy to submit a PR but was just looking for some consensus and confirmation I didn't miss something obvious. @LaZsolt did such a good job putting this library together and they wrote the documentation so I especially was hoping for some confirmation from them before I go about changing things.

LaZsolt, 2024-10-22:
Investigation is in progress...

whyameye, 2024-10-22:
I feel like I must be losing my mind.

I found issue #289 from previous me stating the exact opposite i.e. that the pins do go high when in deep sleep. So I tested another miniEVB and it acted the same as the first: pins were floating not high. Both of my miniEVBs were brand new unused before I tested. Then I tested on a 48 pin version and the pins went high when in deep sleep.

I'm not sure what to say except I'm sorry to be so confused and confusing. An inconsistency in hardware seems unfathomable and yet I don't have another explanation.

LaZsolt, 2024-11-17:
I measured the current which flows through the glowing LED when DPS2 mode is on. It is between 3 and .8 microamps. 
This current much lower than a pull-up resistor allows.
Not every LEDs can glow shuch a small current.
So 100K resistor easily can pull down to GND or pull up to VCC the pin voltage when DPS2 mode active.

Conclusion: The LED's glow is not caused by the internal pull-up resistance, but by other leakage currents come from of VCC.