dbuezas/lgt8fx #327: Arduino as ISP support

KLevi0, 2024-12-05:
Hi! First I'd like to thank you for your work.
This is just a suggestion of an easy way to burn the bootloader. I uploaded [https://github.com/DavidGuo-CS/OSOYOO_Arduino/blob/main/libraries/Lgt328P_ISP/Lgt328P_ISP.ino] on my ATmega328p (Arduino UNO) and then added Arduino as ISP in the _programmers.txt_ file to burn this bootloader. It would be really useful if this was included by default because I spent a lot of time looking for a solution. 
After I added it, it now works flawlessly, much better than the 4MHz prescaler from the factory.
 [image]

dbuezas, 2024-12-06:
Hi!

Is it better than the built in one?
https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/libraries/LarduinoISP/readme.md

KLevi0, 2024-12-06:
I don't know about that, BUT as you can see I couldn't select Arduino as ISP to use that. It was not included in the dropdown menu.

dbuezas, 2024-12-06:
Oh you used an atmel as ISP

dbuezas, 2024-12-06:
AVR as ISP would work

KLevi0, 2024-12-06:
Indeed, it should work but for some reason it doesn't. I tried several times burning the bootloader back the back. Same wiring, same sketch, Arduino as ISP uploaded 100%, while AVR ISP produced this error message
```
avrdude stk500_getsync() warning: attempt 10 of 10: not in sync: resp=0xe0
avrdude main() error: unable to open programmer stk500v1 on port COM9
```
The programmer setup looks similar to me. For ease of modification I just replaced the values for the built in one. It was trial and error haha
```
avrisp.name=AVR ISP
avrisp.communication=serial
avrisp.protocol=stk500v1
avrisp.speed=19200
avrisp.program.tool=avrdude
avrisp.program.protocol=stk500v1
avrisp.program.tool=avrdude
avrisp.program.extra_params=-P{serial.port}

lgtmkii.name=Arduino as ISP
lgtmkii.communication=serial
lgtmkii.protocol=stk500v1
lgtmkii.speed=19200
lgtmkii.program.protocol=stk500v1
lgtmkii.program.speed=19200
lgtmkii.program.tool=avrdude
lgtmkii.program.extra_params=-P{serial.port} -b{program.speed}
```

dbuezas, 2024-12-07:
So the `-b{program.speed}` is missing. But which sketch did burn the atmega328 isp with?

KLevi0, 2024-12-07:
The ISP sketch on the ATmega328 is https://github.com/DavidGuo-CS/OSOYOO_Arduino/blob/main/libraries/Lgt328P_ISP/Lgt328P_ISP.ino .

dbuezas, 2024-12-07:
do you want to make a pr with that link and the corrected programmers.txt?