dbuezas/lgt8fx #328: delay() not working

romoloman, 2024-12-31:
I'm trying to use the delay() function in a program and the delay(500) which is supposed to wait 500 milliseconds return randomly at any delay. I'm using version 2.0.7
I know that delay normally is not the best way to wait as it's supposed to be blocking but in this case it's absolutely correct to block the code.
Here is and extract from my code:
`void setup() {
  pinMode(LEDPIN, OUTPUT);
  pinMode(ALARMPIN, OUTPUT);
  digitalWrite(ALARMPIN, LOW);
  
#ifdef DEBUG
  Serial.begin(115200);
#endif  
  inputServo.attach();  // attaches the servo input interrupt
  while (inputServo.available() == false) {
#ifdef DEBUG
    Serial.println("waiting for RX");
    Serial.println();
#endif      
    delay(500);
    if (ledState == LOW) {
      ledState = HIGH;
    } else {
      ledState = LOW;
    }
    // set the LED with the ledState of the variable:
    digitalWrite(LEDPIN, ledState);
  }`

dbuezas, 2024-12-31:
Delay uses micros(), which depends on timer0. You may be using timer0 for something else.
You can use delayMicroseconds instead but make sure not to go beyond 65535. You can put it in a for loop to reach the total of 500_000

romoloman, 2024-12-31:
millis() is working, if i do a unsigned long now=millis(); while (millis()-now <500) {}. it wait half a second... BTW the same code using an original atmega works...

dbuezas, 2024-12-31:
That's basically how delay is implemented, except that it uses micros instead of millis https://github.com/dbuezas/lgt8fx/blob/c88788c9b062677cddb25aafafc1775ec61e84a4/lgt8f/cores/lgt8f/wiring.c#L112

LaZsolt, 2025-01-01:
@romoloman 
> The problem seems to arise at 32Mhz

In my opinion there is some kind of (compiler) overflow somewhere in the code at this clock frequency.

_In this example_
```
#define clockCyclesPerMicrosecond() ( F_CPU / 1000000L )
#define clockCyclesToMicroseconds(a) ( (a) / clockCyclesPerMicrosecond() ) // == (a) / ( F_CPU / 1000000L ) 
#define microsecondsToClockCycles(a) ( (a) * clockCyclesPerMicrosecond() ) // == (a) * ( F_CPU / 1000000L )
```
_when the value of (a) grater than 134 (microseconds) the compiler at 32 Mhz this calucation overflow:_ **(a) * F_CPU**

I didn't check.

romoloman, 2025-01-02:
I start supposing a hardware bug, the same binary perform correctly in 4 board out of 5 but not on the very first one I have tested.

romoloman, 2025-01-02:
> @romoloman
>
>
> In my opinion there is some kind of (compiler) overflow somewhere in the code at this clock frequency.
>
> _In this example_
>
> ```
> #define clockCyclesPerMicrosecond() ( F_CPU / 1000000L ) [...]

No it's impossible that it generates overflow, first of all is a macro, so overflow depends on the kind of argument.
The only one that possibly can overflow is microsecondsToClockCycles but is used only in pulsein and the argument is unsigned long. 
F_CPU/1000000 = 32 so a*32 indeed it can overflow but the value highly exceed 134 (why 134 ??).

The pulsein function (used by the ServoInput library) has the following definition
unsigned long pulseIn(uint8_t pin, uint8_t state, unsigned long timeout) 
so the compiler should cast any argument give as timeout to unsigned long and then pass it to the macro:
unsigned long maxloops = microsecondsToClockCycles(timeout)/16;
Only if timeout exceed 2^26  the compiler overflows.

The problem in my case was only a defective micro, in more then 40 years this is the very first time I see this sort of issue in a micro, as the micro apparently was working but instead, and depending on the code it was running, it was generating resets and interrupts by it's own.

closed 2025-01-02 by romoloman

dbuezas, 2025-01-03:
Maybe timer0 is kaputt. Which sounds quite ridiculous

hmeijdam, 2025-01-03:
You were running at 32MHz from a very low voltage source? I tried to find in the datasheet what the minimum VCC for 32MHz should be, but failed to find that information in it.