dbuezas/lgt8fx #330: How i should run a LCDMENU in Arduino

mrzfx-96, 2025-01-28:
hello everybody, I'm using this following code from Forntohs website and it doesn't work because of this error pleas help me to solve this problem.

> In file included from C:\Users\Reza\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.7\variants\lgt8fx8p/pins_arduino.h:36:0,
                 from C:\Users\Reza\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.7\cores\lgt8f/Arduino.h:626,
                 from C:\Users\Reza\AppData\Local\arduino\sketches\DAF94D33E26C690D96FDC586EC29A7BB\sketch\sketch_jan28a.ino.cpp:1:
c:\users\reza\appdata\local\arduino15\packages\lgt8fx\hardware\avr\2.0.7\variants\standard\pins_arduino.h:126:12: error: expected identifier before numeric constant
 #define C0 14
            ^
c:\Users\Reza\Documents\Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h:54:9: note: in expansion of macro 'C0'
         C0,
         ^~
c:\users\reza\appdata\local\arduino15\packages\lgt8fx\hardware\avr\2.0.7\variants\standard\pins_arduino.h:126:12: error: expected '}' before numeric constant
 #define C0 14
            ^
c:\Users\Reza\Documents\Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h:54:9: note: in expansion of macro 'C0'
         C0,
         ^~
c:\users\reza\appdata\local\arduino15\packages\lgt8fx\hardware\avr\2.0.7\variants\standard\pins_arduino.h:126:12: error: expected unqualified-id before numeric constant
 #define C0 14
            ^
c:\Users\Reza\Documents\Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h:54:9: note: in expansion of macro 'C0'
         C0,
         ^~
In file included from C:\Users\Reza\AppData\Local\Temp\.arduinoIDE-unsaved2025028-8428-znllj3.a6n4\sketch_jan28a\sketch_jan28a.ino:4:0:
c:\Users\Reza\Documents\Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h:95:5: error: 'CodeSet' does not name a type
     CodeSet codeSet = CodeSet::C0;
     ^~~~~~~
c:\Users\Reza\Documents\Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h: In function 'void reset()':
c:\Users\Reza\Documents\Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h:119:9: error: 'codeSet' was not declared in this scope
         codeSet = CodeSet::C0;
         ^~~~~~~
c:\Users\Reza\Documents\Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h:119:19: error: 'CodeSet' has not been declared
         codeSet = CodeSet::C0;
                   ^~~~~~~
In file included from C:\Users\Reza\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.7\variants\lgt8fx8p/pins_arduino.h:36:0,
                 from C:\Users\Reza\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.7\cores\lgt8f/Arduino.h:626,
                 from C:\Users\Reza\AppData\Local\arduino\sketches\DAF94D33E26C690D96FDC586EC29A7BB\sketch\sketch_jan28a.ino.cpp:1:
c:\users\reza\appdata\local\arduino15\packages\lgt8fx\hardware\avr\2.0.7\variants\standard\pins_arduino.h:126:12: error: expected unqualified-id before numeric constant
 #define C0 14
            ^
c:\Users\Reza\Documents\Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h:119:28: note: in expansion of macro 'C0'
         codeSet = CodeSet::C0;
                            ^~
In file included from C:\Users\Reza\AppData\Local\Temp\.arduinoIDE-unsaved2025028-8428-znllj3.a6n4\sketch_jan28a\sketch_jan28a.ino:4:0:
c:\Users\Reza\Documents\Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h: In function 'void handleIdle()':
c:\Users\Reza\Documents\Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h:140:17: error: 'menu' was not declared in this scope
                 menu->process(ENTER);
                 ^~~~
c:\Users\Reza\Documents\Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h:140:17: note: suggested alternative: 'min'
                 menu->process(ENTER);
                 ^~~~
                 min
c:\Users\Reza\Documents\Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h: In function 'void handleReceived(unsigned char)':
c:\Users\Reza\Documents\Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h:153:17: error: 'codeSet' was not declared in this scope
         switch (codeSet) {
                 ^~~~~~~
c:\Users\Reza\Documents\Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h:154:18: error: 'CodeSet' has not been declared
             case CodeSet::C0:
                  ^~~~~~~
In file included from C:\Users\Reza\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.7\variants\lgt8fx8p/pins_arduino.h:36:0,
                 from C:\Users\Reza\AppData\Local\Arduino15\packages\lgt8fx\hardware\avr\2.0.7\cores\lgt8f/Arduino.h:626,
                 from C:\Users\Reza\AppData\Local\arduino\sketches\DAF94D33E26C690D96FDC586EC29A7BB\sketch\sketch_jan28a.ino.cpp:1:
c:\users\reza\appdata\local\arduino15\packages\lgt8fx\hardware\avr\2.0.7\variants\standard\pins_arduino.h:126:12: error: expected unqualified-id before numeric constant
 #define C0 14
            ^
c:\Users\Reza\Documents\Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h:154:27: note: in expansion of macro 'C0'
             case CodeSet::C0:
                           ^~
c:\users\reza\appdata\local\arduino15\packages\lgt8fx\hardware\avr\2.0.7\variants\standard\pins_arduino.h:126:12: error: expected ':' before numeric constant
 #define C0 14
[... 64 lines cut ...]
// Initialize the main menu items
// clang-format off
MENU_SCREEN(mainScreen, mainItems,
    ITEM_BASIC("Start service"),
    ITEM_BASIC("Connect to WiFi"),
    ITEM_BASIC("Settings"),
    ITEM_BASIC("Blink SOS"),
    ITEM_BASIC("Blink random"));
// clang-format on

LiquidCrystal_I2C lcd(0x27, LCD_COLS, LCD_ROWS);
CharacterDisplayRenderer renderer(new LiquidCrystal_I2CAdapter(&lcd), LCD_COLS, LCD_ROWS);
LcdMenu menu(renderer);
KeyboardAdapter keyboard(&menu, &Serial);

void setup() {
    Serial.begin(9600);
    // Initialize LcdMenu with the menu items
    renderer.begin();
    menu.setScreen(mainScreen);
}

void loop() {
    keyboard.observe();
}`

LaZsolt, 2025-01-28:
In the Arduino library `Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h` is not compatible with LGT8F328 board definitions because LGT8F port definiton macros `C0`, `C1`, `C2` using the same name as in `KeyboardAdapter.h`

mrzfx-96, 2025-01-28:
> In the Arduino library `Arduino\libraries\LcdMenu\src/input/KeyboardAdapter.h` is not compatible with LGT8F328 board definitions because LGT8F port definiton macros `C0`, `C1`, `C2` using the same name as in `KeyboardAdapter.h`

so how can i fix it? should i use another board? or this error is solvable?

lalo-uy, 2025-01-28:
You can change the enum in KeyboardAdapter from C0 to C0_Cod or something.
Also C1.
Check all ocurences in the file.

mrzfx-96, 2025-01-28:
``

> C0_Cod

I did, But it Doesn't work yet, I have same error.

LaZsolt, 2025-01-31:
First make a backup copy of this file, 

`c:\users\reza\appdata\local\arduino15\packages\lgt8fx\hardware\avr\2.0.7\variants\standard\pins_arduino.h`

after that remove this lines:

https://github.com/dbuezas/lgt8fx/blob/c88788c9b062677cddb25aafafc1775ec61e84a4/lgt8f/variants/standard/pins_arduino.h#L125-L132

Try to compile.