dbuezas/lgt8fx #332: uploading problem with pl2303

Amoodoge, 2025-04-20:
hi guys i have a problem that i cant upload any sketches to my lgt8f328p lqfp32 with my pl2303 usb to ttl.
 when i uploaded blink on it, it uploaded without problem and its now running blink perfectly and seems anything work!! then when i wanted to upload another blink on lgt it gave me this error 

`Arduino: 1.8.19 (Windows 10), Board: "LGT8F328, 64 (normal), Internal 32MHz, 1, 328P-LQFP32 (e.g. MiniEVB nano-style or WAVGAT), 57600"

Sketch uses 1114 bytes (3%) of program storage space. Maximum is 29696 bytes.

Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x35

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x35

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x35

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x35

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x35

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x35

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x35

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x35

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x35

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x35

Problem uploading to board.  See https://support.arduino.cc/hc/en-us/sections/360003198300 for suggestions.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`
I tried everything! I did check my drivers and changed the uploading speed and clock divider and I still have this problem.
I guess my programmer is malfunctioning. 
Please help

LaZsolt, 2025-04-21:
Have you tried this solution?

https://www.ifamilysoftware.com/Prolific_PL-2303_Code_10_Fix.html

Amoodoge, 2025-04-21:
Yes i did that before and
i can see Prolific USB-to-Serial Comm Port (COM9) in device manager

but still i see this freak error

zielony12, 2025-09-28:
if you are using arduino ide try setting programmer to usbasp and set the board to arduino micro and choose old bootloader variant. press reset button after clicking upload if you don't have data ready pin from the programmer

zielony12, 2025-09-28:
i set the programmer to lgtswd mkll as isp choose the variant and it works but not every time maybe because i havent soldered the pins. double check the serial pinout. im on linux btw