dbuezas/lgt8fx #333: analogWrite does not work with D9 and D10

Tivaty, 2025-05-23:
Hello,

Can you check if there is any problem with D9 D10 when using analogWrite?

I tested all the pin with 2 miniEVB and all of digital pins working ok with analogWrite even pins without PWM (work like on - off when increase analogWrite paratmer);

D3, D5, D6, D11 work ok at all. but only D9 and D10 just can work with digitalWrite.

Hope to here back from you.

Thank you

hmeijdam, 2025-05-23:
Must be something else wrong on your end

This sketch works fine on my pseudo promini EVB
```
const int lowestPin = 9;
const int highestPin = 10;

void setup() {
  // set pins 2 through 13 as outputs:
  for (int thisPin = lowestPin; thisPin <= highestPin; thisPin++) {
    pinMode(thisPin, OUTPUT);
  }
}

void loop() {
  // iterate over the pins:
  for (int thisPin = lowestPin; thisPin <= highestPin; thisPin++) {
    // fade the LED on thisPin from off to brightest:
    for (int brightness = 0; brightness < 255; brightness++) {
      analogWrite(thisPin, brightness);
      delay(2);
    }
    // fade the LED on thisPin from brightest to off:
    for (int brightness = 255; brightness >= 0; brightness--) {
      analogWrite(thisPin, brightness);
      delay(2);
    }
    // pause between LEDs:
    delay(100);
  }
}
```
Adapted from the "Mega analogWrite() test" example that is part of the Arduino IDE.

jayzakk, 2025-05-23:
I am not sure if I looked at the correct pinout, but isn't the miniEVB with a combined pin 8+9 and 10+11? I don't have such device, so can't tell for sure.
If so, depending on what is the default, only one of the double used pins should be used / maybe the "do not want pin" even needs to get disabled? D8 is not pwm, D9-D11 are. 
Crazy pinout!😂

jayzakk, 2025-05-23:
> [edit] However I do see a problem with PD2. It responds to analogWrite as a digital-only pin.

That's because PD2 is not a PWM pin.

jayzakk, 2025-05-23:
> > [edit] However I do see a problem with PD2. It responds to analogWrite as a digital-only pin.
> 
> That's because PD2 is not a PWM pin on SSOP20 devices. At least Google tells me. But why?

Answering myself, as I am only using the nano style boards, i never thought about:
D2 is PD2+PF2. Same for D1.
The two F port pins can do PWM via oc3a and oc3b. So it never was PD2 doing the PWM. Yay.

hmeijdam, 2025-05-23:
I tested PD2 on this board. According the pinout diagram PD2 is the pin with PWM. I am with you about OC3A and OC3B, so maybe in the pinout diagram the PF2 should be listed as PF2~ and the D2~  should be listed as D2.

[image]

But even then PF2 does not want to do PWM either, as with this sketch I only get a blink.

```
#define LEDPIN PF2

void setup() {
  pinMode(LEDPIN, OUTPUT);
}

void loop() {
  for (int brightness = 0; brightness < 255; brightness++) {
    analogWrite(LEDPIN, brightness);
    delay(2);
  }
  for (int brightness = 255; brightness >= 0; brightness--) {
    analogWrite(LEDPIN, brightness);
    delay(2);
  }
  delay(100);
}

```

And if I select the non existing PF6 pin as LEDPIN I get PWM on D6~ . So I suspect all PF% pins are 1:1 translated into PD% pins and then processed by analogWrite().

For the PWM to work on PF2 I need to use registers directly
```

void setup() {
  TCCR3A = 1 << COM3B1 | 1 << WGM30;
  TCCR3B = 1 << WGM32  | 1 << CS30;
  DDRF =   1 << DDF2;
}

void loop() {
  for (int brightness = 0; brightness < 255; brightness++) {
    OCR3B = brightness;
    delay(2);
  }
  for (int brightness = 255; brightness >= 0; brightness--) {
    OCR3B = brightness;
    delay(2);
  }
  delay(100);
}
```

Tivaty, 2025-05-23:
> I tested PD2 on this board. According the pinout diagram PD2 is the pin with PWM. I am with you about OC3A and OC3B, so maybe in the pinout diagram the PF2 should be listed as PF2~ and the D2~ [...]

closed 2025-05-23 by Tivaty

jayzakk, 2025-05-23:
Ok, thats not the board I was thinking of.

I just set up some nano style board with test code for analogWrite on {1, 2, 3, 5, 6, 9, 10, 11}
But, D9 does work perfectly for me. So this is not related to your issue.

Anyways, D1 and D2 are not doing PWM as they should. Looking into that one at least. All of timer3 references and definitions are missing for analogWrite :-(