dbuezas/lgt8fx #336: Sleep functions cause reset in code execution after waking up

Trotyl64, 2025-07-13:
I'm using the pro mini style board with nothing connected and the powerDownWakePeriodic example.

The sleep function works but right after waking up the microcontroller "resets" and goes back to executing code from setup() instead of continuing to the rest of the code. What's interesting is there's no fast blink that normally happens when resetting with the button. I guess it has something to do with the reset vector and WDT mentioned in another issue.

Trotyl64, 2025-07-14:
After more investigation I found that there's two things causing reset - wrong watchdog timer settings and BOD. I've been fiddling with the powerDown function for the LGT8FX8P and it seems that the WTD is set improperly, specifically the WDE pin is set to 1, causing the WDT to run in "interrupt reset" mode instead of the normal interrupt mode. I replaced the WDT enable code with one from the PMU.cpp enable_wdt function and it works properly,
`uint8_t btmp;

asm ("wdr");

// disable WDE
btmp = (WDTCSR | 0x90) & 0xF7;
WDTCSR = 0x18 | btmp;
WDTCSR &= 0xF7;

// enable wdt
btmp = period & 0x7;	// WPD[2:0]
if(period > SLEEP_8S)
  btmp |= 0x20;	// WPD[3]

btmp |= 0x40;	// WDIE enable
btmp |= 0x10;	// WDCE enable
WDTCSR = 0x18 | btmp;
WDTCSR = btmp;`

By trial and error I found that this line also somehow causes the MCU to reset `if (bod == BOD_OFF) s_unlockWrite(VDTCR,o_vdtcr);`
I've found that using sleep with BOD_ON parameter doesn't fix this problem, only after commenting this line(and replacing the WDT code) the MCU wakes up from sleep and continues the code.

For debugging purposes I had a Serial.println and a blink code right after the sleep function and I later found out that unless the following serial functions: begin(9600), println(), end() are right after the powerDown() the MCU will not wake up properly. Here's the loop() code that works: `void loop() 
{
    LowPower.powerDown(SLEEP_4S, ADC_OFF, BOD_OFF);
    Serial.begin(9600);
    Serial.println("Successfully woke up!");
    Serial.end();
    digitalWrite(LED_BUILTIN, HIGH);
    delay(5000);
    digitalWrite(LED_BUILTIN, LOW);
}`
With this revelation I'm completely lost as to what could be the cause, I currently have only one pro mini style board so I can't verify if this particular one is broken or not.

LaZsolt, 2025-07-14:
Hello!

LGT8F328P sleep modes are very different compared to AtMega.

Read this post: https://wolles-elektronikkiste.de/en/minievb-boards-an-overview At the end of the post (chapter: Power consumption in sleep mode) you will find how to use LGT8F328P sleep modes. 
And read the library document: https://github.com/dbuezas/lgt8fx/tree/master/lgt8f/libraries/lgt_LowPower

Trotyl64, 2025-07-15:
Hello, thanks for the help!
I've read both the post and the document, however I still have no idea why even the examples don't work how they're supposed to, am I really doing something wrong? Neither **powerDownWakePeriodic.ino** nor the **powerDown-RestoreWDT.ino** examples work properly. 

The following code is the same as **powerDownWakePeriodic.ino** only with serial added. It only ever prints "Sleeping", completely skipping the second part. The LED also doesn't light up besides an extremely short blink when waking up.
`#include "lgt_LowPower.h"

void setup()
{
  // Pin settings for low consumption in sleep mode.
    DDRB  = 0x00;                 // Set all pins to input
    DDRC  = 0x00;
    DDRD  = 0x00;
    DDRE  = 0x00;
    PORTB = 0xFF;                 // Set all input with internal pull up resistors
    PORTC = 0xFF;
    PORTD = 0xFF;
    PORTE = 0xFF;
    pinMode(LED_BUILTIN, OUTPUT); // Visual effect to show program status
    digitalWrite(LED_BUILTIN, LOW);
    VDTCR = 0x80;
    VDTCR = 0x4B;                 // LVD (BOD) enable to 2.5 V for demonstration purpose 
}

void loop() 
{
    delay(2000);
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, LOW);

    Serial.begin(9600);
    Serial.println("Sleeping");
    Serial.end();

    // Enter power down state for 8 s with ADC and BOD (LVD) modules disabled
    LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
    
    // Do something here
    // Example: Read sensor, data logging, data transmission.
    Serial.begin(9600);
    Serial.println("Executing code");
    Serial.end();
    digitalWrite(LED_BUILTIN, HIGH);
}`

LaZsolt, 2025-07-15:
Sorry, I have no time to build a test circuit in the near future...

I noticed that after writing to the serial device, program immediately starts sleeping without waiting for the end of the data transfer.
So I think need to wait with this: ``` Serial.flush(); ``` This function waits for the transmission of outgoing serial data to complete.

```
    Serial.begin(9600);
    Serial.println("Sleeping");
    Serial.flush();
    Serial.end();
```
`Serial.end();`  will probably also wait for the outgoing data transmission to complete, but the Arduino language ducumentation not mention it.

Another thing: **Maybe** LowPower library has a B U G. A big bug.
The LowPower library uses the watchdog for timing, but the watchdog interrupt not handled. So first of all 

```
#include "WDT.h" 
#include "lgt_LowPower.h"

volatile boolean isrflag;
```

and try handling the WDT interrupt like in the example mentioned below.
```
ISR (WDT_vect)
{
    isrflag = true;
    wdt_reset();
}
```

https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/libraries/WDT/examples/wdt_interrupt/wdt_interrupt.ino

I hope this will be useful.

Trotyl64, 2025-07-16:
Sorry but I'm not entirely sure how to incorporate sleep into this example, I wasn't able to get it to work.

I noticed that including WDT.h changes the behavior of the code below.
Without WDT `i` is incremented by 2, suggesting that code after sleep is executed(although the LED is not turned on for some reason).

With WDT included `i` is only incremented by 1 and sometimes prints jibberish.

With BOD_OFF it only prints "Setup" and "Sleeping, i: 0" all the time, this happens regardless if WDT.h is included or not. I mentioned this in an earlier comment that the code for reenabling BOD in powerDown() seems to reset the microcontroller.
[To better explain what I mean here are screenshots from the serial monitor in arduino IDE](https://imgur.com/a/iw1lykG)
I don't know much about the low level parts of a microcontroller, sorry if I'm doing something wrong.

```#include "lgt_LowPower.h"
// #include "WDT.h"

int i = 0;
void setup()
{
  // Pin settings for low consumption in sleep mode.
    DDRB  = 0x00;                 // Set all pins to input
    DDRC  = 0x00;
    DDRD  = 0x00;
    DDRE  = 0x00;
    PORTB = 0xFF;                 // Set all input with internal pull up resistors
    PORTC = 0xFF;
    PORTD = 0xFF;
    PORTE = 0xFF;
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, LOW);
    VDTCR = 0x80;
    VDTCR = 0x4B;                 // LVD (BOD) enable to 2.5 V for demonstration purpose 

    delay(2000);
    Serial.begin(9600);
    Serial.println("Setup");
    Serial.flush();
    Serial.end();
}

void loop() 
{
    digitalWrite(LED_BUILTIN, LOW);

    Serial.begin(9600);
    Serial.print("Sleeping, i: ");
    Serial.flush();
    Serial.println(i);
    Serial.flush();
    Serial.end();
    i++;
    if(i >= 4)
    {
        Serial.begin(9600);
        Serial.println("test");
        Serial.flush();
        Serial.end();
        // i = 0; //uncommenting this line breaks the whole thing, serial prints jibberish
        //digitalWrite(LED_BUILTIN, HIGH); //uncommenting this line also breaks the whole thing
    }

    LowPower.powerDown(SLEEP_2S, ADC_OFF, BOD_ON);
    
    i++;
    digitalWrite(LED_BUILTIN, HIGH);//The LED doesn't turn on for some reason
}